Snapmaker/Snapmaker2-Controller #142: Feature request:  M00 / M01 gcode support

smhoover, 2021-09-15:
Hello team! 

I'm interested in using the M00 code for the cnc module and I'm not sure if it it working correctly or fully implemented. 

 I see in this document https://snapmaker.github.io/Documentation/gcode/M000-M001  that the mcode might be supported but I cannot get the machine to use pause or resume from luban. I hard code in the m00 or m01 code in my  file but the controller acts as if the code is not present or freezes requiring a hard restart. 

Can you please look into this?

Best regards 

Can you please look into this for me?

xhycode, 2021-09-16:
Hi  smhoover,
  Sorry for the confusion caused by our documentation not being up to date, snapmaker2.0 do not support M0/M1 commands.
  Can you tell me what you need to do so, maybe I can try to help you in another way.

smhoover, 2021-09-16:
Absolutely.

I have written some gcode to engrave numbers on 20 sided die and I would
like to have an optional stop that will allow me to pause so I can rotate
the faces to mark all faces.

 Normally, I would code in a M00 code for this.

Any help would be greatly appreciated. Possibly using the new filament
change code?

smhoover, 2021-09-16:
> Hi smhoover.
> If it is 3D printing you can use M600 pause, I'm sorry that currently the
> laser and CNC do not support the command on GCODE pause.
> Have communicated with the product manager can add this command, because
> now is developing other modules, until the official release of the firmware
> or a long time.
> If you are in a hurry, we can provide a separate test firmware for you to
> use during the testing phase. Even then it won't be available soon, because
> it needs to be developed in conjunction with the screen to be implemented.
> We are sorry that we cannot help you immediately with your problem.

I understand, and I would love to help test when firmware is available.

Would a tool change command possibly work?

xhycode, 2021-09-17:
Work is controlled by the screen, the current controller can not directly pause the screen work, we will add a new protocol to support this feature

uefvanscheld, 2022-07-03:
Hi,
found this thread since I struggeling with tool changes too.
Any updates of providing the M0 command for CNC ?

FoTo50, 2023-10-10:
Hi,
I also would love to see this specially for tool changing, so hoping to get updated from here soon ;)