Snapmaker/Snapmaker2-Controller #143: Changing the co-ordinates for bed levelling

zk100005, 2021-09-15:
Hi,
I am using A150 bed on A350 machine. I am trying to change the bed area by changing the probe area but nothing seems towork for me. please help. Even chanigng the X_MAX_POS doesnt seem to have any effect on the maximum distance that I can move the printhead on x-axis. Seems like any change I am making to Marlin/configuration.h doesnt seem to reflect on the machine

artemician, 2021-09-15:
Why are you using a150 bed on a350? That's the bigger question..

xhycode, 2021-09-16:
Hi  zk100005,
The code is in [Snapmaker2-Controller/snapmaker/src/module/linear.cpp](https://github.com/Snapmaker/Snapmaker2-Controller/blob/ee368b40949e334bd354314fd33356bfd7df1117/snapmaker/src/module/linear.cpp#L398) assignment of A350 length
Can you tell us why you need to connect the A150 hot bed to the A350?

zk100005, 2021-09-16:
Actually I want to have a smaller print area but I want ot do some imaging of the printhead on the extra length left from the x-axis therefore I am trying to attempt this. However, My problem is that no change that I am making to the firmware controller is getting reflected on the machine.

zk100005, 2021-09-16:
And what is the difference between X_DEF_SIZE and X_MAX_POS

xhycode, 2021-09-16:
X_DEF_SIZE is the size of the heat bed, used for leveling, you can see the grid size on the heat bed.
X_MAX_POS is the travel size of the axis

zk100005, 2021-09-16:
Thanks a lot :D Let me check if changing this does anything to the machine. Do you think it could be possible that my original firmware is not getting overridden by the changes that I am making

zk100005, 2021-09-16:
> X_DEF_SIZE is the size of the heat bed, used for leveling, you can see the grid size on the heat bed.
> X_MAX_POS is the travel size of the axis

Are there any EEPROM settings that might be hindering the changes that I am making?

xhycode, 2021-09-16:
Change the SHORT_BUILD_VERSION version number in Version.h after each modification, the same version number controller will not be updated

zk100005, 2021-09-16:
> Change the SHORT_BUILD_VERSION version number in Version.h after each modification, the same version number controller will not be updated

Thanks a ton !
I'll get back to you

scyto, 2021-09-17:
I would like to see the bed levelling process actually measure further to the right edge, left edge and back (front seems ok) to create a more accurate leveling mesh around the edges.

zk100005, 2021-09-22:
> Hi zk100005,
> The code is in [...]

Could you tell me how to change the points for levelling ?
The bed size I changed , but it didnt affect much. Because the axis lengths are varying here. Can you tell me how I can vary the points for auto levelling or manual levelling? whichever should work