Snapmaker/Snapmaker2-Controller #147: Can we have a button to move files from USB drive to the machine's internal storage and vice versa?

NomadicVoxel, 2021-09-17:
The fact that you can send files from Luban to the Snapmaker 2 demonstrates that the S2 is capable of storing files internally, but there is no other way to access this besides Luban, even on the machine itself. It's _technically_ not a problem, but I would greatly appreciate being able to move files from the flashdrive to internal storage and vise versa. 

The functionality's pretty much *there* already. Just that there's not a button for it. Gcode is transferred to internal memory when it's preparing to execute it. It's in long-term storage, evidenced by the blackout recovery features. All we need is someone to add in a button for it, maybe in that file menu you get by dragging a file's tile? Right by Delete and Rename.

NomadicVoxel, 2021-09-17:
Maybe something like this for the icon 
[image]

NomadicVoxel, 2021-09-17:
// crap didn't mean to close it

CIPop, 2021-10-07:
+1 on this - initially I thought it's not there to somehow protect the internal flash but the machine always transfers from USB to local. 

It would be nice to batch transfer a few files to internal to avoid leaving the USB plugged in for the next prints. 

---

Workaround (not really usable for batch copy): my machine has an "Experimental" button I found that there is a "Copy test" function that will copy from USB to local but it will rename the file to testCopy.gcode. I can then rename that file to the original filename and do this again.

NomadicVoxel, 2021-10-09:
Funny, I didn't see experiments on mine, I should look again. Renaming the file to testCopy on transfer is peculiar, but it'll be useful in the meantime, thanks for sharing that.