Snapmaker/Snapmaker2-Controller #151: Large gcode file does not load

siberianhuskies, 2021-10-05:
Hi, I noticed to similar/connected problems with large gcode files.

I'm trying to print a gcode file with about 300 MB from the USB stick. When selecting the file for printing, the "parsing file" screen gets stuck at around 88%. Only option is to manually power off the machine.

A gcode file with about 245 MB, sometimes goes beyond the "parsing file" screen and I can select "start" printing, but it then immediately crashes and after a few seconds I get the error FabScreen not responding. Sometimes the "copying file to touch screen" crashes, too. Again, the only option is to manually power off the machine.

Other much smaller files work with no problem.

Essentially we cannot print any gcode file with more than ~240MB (which is very annoying and sad for an owner of the A350).

I'm not sure what causes this problem, but I assume that the gcode file is maybe first read completely in memory and thus results in the crash because of low memory. If that is indeed the case, than this problem should be fixable: For the parsing (I assume calculating print time etc), there should be no need to keep the complete file in memory and can be parsed line by line. Similar for printing, after copying the file to the internal memory, the file could be read line-by-line and maybe only 100 or 1000 lines of the gcode file could be buffered.
If there are disadvanteges of this approach, than maybe there could be at least a workaround like: If file_size < 240 MB then load_to_memory, else read_line_by_line . So that large gcode files are at least printable.

MlvPerry, 2021-10-05:
Yes, that is a very bad and relevant issue, it can be easily reproduced too. I reported it around mid-January, but nothing has been done about it, it's also CLEARLY not a Luban problem as I stated(!!!).
Hopefully it will get solved soon!

Andy250292, 2022-02-02:
Same problem on the A250. I sliced the files with Simplify3D. The problem is happening when the files are mostly bigger than ~10mb.

ch0t4nk, 2022-09-14:
The problem still exists on A350T