Snapmaker/Snapmaker2-Controller #170: Files sent via WiFi do not load as expected – loading and sorting

jenkins007, 2021-11-11:
When sending a file in various modes of the Snapmaker UI, the new file does not show up. I have to exit out of the "Files" UI and go back to the "Start" UI. Then selecting the files Icon shows the new uploaded file. 

Also, the sorting appears to get adjusted at various times, including power off. I am always wanting the newest file received on top (sort by date). I rarely use the sort by name, but is is useful when looking for an older gcode file.

So this is a 2 part issue:
1. Loading new files requires going back to the "Start" UI before showing up in the "Files" UI.
2. Default sorting of file names needs to be static, i.e. set it once and it will not change unless manually changed – even after power off.

jrntrfs, 2021-11-13:
I’ve noticed that when after recieving, it will be at the bottom of the list, by opening sort and pressing “sort by name” the new file will appear on top.

viper0078, 2021-12-10:
I also use the sort by date.
It is very inconvenient to always have the sort by file name selected. Please change it so that it remembers the last sort order used.
I hope this will be fixed as soon as possible.

spitzlbergerj, 2021-12-30:
yes, please set the sorting back to the default before the last update. Until the update the newest files were always on top, so sorted by date descending

Eddie-Hartman, 2022-01-02:
Defaulting to sort by date would be a possible option or having a setting where we can set what sort we want or remembering what sort option was last used by the user.

I've ran into this weird issue as well where files are just not in any order I'd expect by default and there are many ways to fix this using any of the suggestions above. ^

jrntrfs, 2022-01-03:
> Defaulting to sort by date would be a possible option or having a setting where we can set what sort we want or remembering what sort option was last used by the user.
>
> I've ran into this weird [...]

I don’t think most of you are aware of this. It still is ordered by date, only the files that were allready on the device are above the newly send files. Both groups are ordered by date. Only by re-pressing the order by date option they get arragend correctly. So they’re not random nor alfabetical but somehow the group of previous files get (even though they have an older date)are placed above the newest

Eddie-Hartman, 2022-01-03:
@jrntrfs What you're describing is reverse by date. This isn't exactly the behavior I've seen either. If what you're describing were true, if you hit the order by date button twice, it should be the same order as if you opened it and there as "no" sort at all. This isn't exactly what I have experienced. It's also just odd that it was changed.

I'm more so asking to let us control it rather than change a behavior in an update.

jenkins007, 2022-03-10:
What jrntrfs is stating is that the files are organized by date (in my case descending with newest on top). When new files are sent to the Snapmaker, they are placed at the bottom even though they are the latest date/time. The rest of the files are still sorted by date with the newest on top except for the one that was just sent. Clicking on "Sort by Date" makes the newest one on top along with the other newer ones that were already on top. That is the bug!