Snapmaker/Snapmaker2-Controller #175: Fix Request: New linear module could damage machines

blockmodule, 2021-11-28:
I would like to submit a fix request to GitHub with the following information.

If you build a machine with a new linear module (Read8mm/20mm, stamped on the side) in a combination other than the one shown below, nothing will appear on the controller display screen. If you build your machine with the new linear module (Read8mm/20mm, stamped on the side) in any other combination than the one below, nothing will show up on the controller display screen, and the axis will move on one side during homing, possibly damaging the two rails.

Correct configuration
X-axis: 1 x 20mm
Y-axis: 2 x 20mm
Z-axis: 2 x 8mm

Wrong configuration will cause problems.
X-axis: 1 x 20 mm
Y-axis: 1 x 20 mm 1 x 8 mm
Z-axis: 1 x 20 mm 1 x 8 mm

As a countermeasure, if the wrong combination is made, an error should be displayed at the initial homing point and any operation in jog mode for each axis should not be accepted.

This problem occurs when replacing new models and linear modules.

Snapmaker should include the correct axis configuration in the product package and interlock it in the machine controller firmware.

It should be recognized that this is a serious matter by Snapmaker for safety and machine integrity.

xhycode, 2021-12-13:
Hi blockmodule
This security vulnerability has been addressed in the [just released firmware](https://forum.snapmaker.com/t/snapmaker-2-0-firmware-updates-and-downloads/5443/18). Incorrect model recognition will disable movement.
Thanks again for the heads up!

blockmodule, 2021-12-13:
Hi xhycode !

Thanks for addressing this issue!

I appreciate it.

closed 2021-12-13 by blockmodule