Snapmaker/Snapmaker2-Controller #178: Allow setting work origin permanently to 0/0 instead of it defaulting in the center of the bed

prueker, 2021-11-30:
It would be nice if there was an option for the laser usage to permanently have the work origin start at X 0 and Y 0 instead of it always starting in the center of the plate and then manually having to jog the head to 0/0 when I have used a file like using the camera capture or having prepared it with another software that can use absolute 0,0 based coordinates.

xhycode, 2021-11-30:
Hello prueker
Your suggestion is a good one and I will provide it to our product manager.
You can insert G53 in the gcode file to print using the machine (0,0) based coordinates..
Although this is not convenient, I hope it can temporarily solve your problem.

; G-code START <<<
G53  ;  here
G90
G21
M106 P0 S255
G0 F2500
G1 F2500
G1 X4.78 Y2.41

brent113, 2021-11-30:
I would prefer the Laser and CNC workflows have the option to not reset the work origin.

In normal machine function G54 work coordinate offsets are written to EEPROM and loaded at startup. But the job starting process then resets the work origin to the default location.

If the job starting process had the option to use the previously saved work origin that would greatly speed up several workflows. For example, running a roughing CNC pass followed by a finish pass. It would be nice if the touchscreen and Luban would at least honor the X/Y origin, with some additional functionality to allow setting the new tool height for CNC without crashing into the workpiece.

By allowing remembering a previous origin that would also resolve the original feature request here, as the origin could be set to 0,0 and then all subsequent laser runs could use the previously saved origin.

siberianhuskies, 2022-02-03:
> I would prefer the Laser and CNC workflows have the option to not reset the work origin.
> 
> In normal machine function G54 work coordinate offsets are written to EEPROM and loaded at startup. But the job starting process then resets the work origin to the default location.
> 
> If the job starting process had the option to use the previously saved work origin that would greatly speed up several workflows. It would be nice if the touchscreen and Luban would at least honor the X/Y origin.
> 
> By allowing remembering a previous origin that would also resolve the original feature request here, as the origin could be set to 0,0 and then all subsequent laser runs could use the previously saved origin.

I totally agree with brent113. This is especially useful for laser engraving/cutting multiple times the "same" object, for example multiple coasters. Or repeating an engraving, because the burn was not "black" enough.

pauserratgutierrez, 2022-03-07:
I agree with this topic, the center of the bed isn't and shouldn't be (0,0) but (maxX - minX, maxY - minY)