Snapmaker/Snapmaker2-Controller #184: Stop button should not 'crash' SnapMaker.

kalmdown, 2021-12-09:
The SM should pause the current position of the head (and possibly move up vertically a few MM) and give the user options - pause (user will move obstruction), stop (whole setup needs to be rethought), settings (problem can be solved by moving the origin) - to recover.

brent113, 2021-12-09:
If the developers would prefer to allow users to perform a hard emergency shutdown of the system for maximum safety, perhaps an option could be given to users to make the e-stop button behave more like the pause function on the screen, with a disclaimer if necessary.

Gb1942, 2021-12-09:
And keep the lights on!!

kalmdown, 2021-12-09:
It is not doing a hard shutdown (ie killing the power) with the current implementation, the SM was still On, so its not like electronics were protected or physical parts would be stopped from [I don't know...being absurd?] catching fire. 99% of the point of the Stop Button is stop the machine from moving in a direction which will damage the head. Just stopping movement is good enough. After that, the user should be left in a safe position with the ability to evaluate the situation.

Gb1942, 2021-12-09:
So no problem to keep the lights on.

bf8392, 2022-08-04:
+1 I would suggest leave it as a hard stop, but without the necessity of reboot if the knob is turned upwards again, you can give a screen confirm that the issue is resolved, and than "exit to homescreen"

kalmdown, 2022-08-04:
Prefer to keep lights on...to review situation.

The3rdIcon, 2022-12-16:
The current behavior really needs to change as it is against expectation of most people. On other machines the emergency stop is more like a pause to prevent damage.

caerandir, 2023-04-16:
I also would like to have a way to resume a stopped print. The suggested way of making it the user's choice if it is a hard stop or a softer one would be neat. For the hard stop, it would be great if like with power loss resume you could also resume the stopped job.