Snapmaker/Snapmaker2-Controller #190: Can the firmware safely be downgraded after installing 4.4.11

brent113, 2021-12-14:
In the past if the new firmware caused unexpected issues we have been able to downgrade to the previous version.

Also I understand that the touchscreen firmware cannot be downgraded as that is a limitation of Android

With this version changing the touchscreen protocol will there be expected issues downgrading the controller to the previous firmware but leaving the touchscreen on the latest app?

xhycode, 2021-12-15:
The screen is not restricted for downgrading
Can you describe the procedure and provide information about the version you downgraded?
Did you encounter any problems that required downgrading?

brent113, 2021-12-15:
Thanks for the quick response.

> Did you encounter any problems that required downgrading?

I'm trying to assist other people on the forum and I want to avoid giving bad advice so I want to understand how downgrading works.

For example, today these threads:
https://forum.snapmaker.com/t/latest-firmware-issue/22666/4
https://forum.snapmaker.com/t/1-16-0-firmware-laser-issue/22674
https://forum.snapmaker.com/t/set-safe-work-origin-after-firmware-update-if-you-dont-want-your-snapmaker-destroyed-that-is/22641

And a couple of others.

In the past when issues like this have happened a common troubleshooting step is to see if the previous firmware version would fix the behavior to see if it's limited in effect to the firmware version recently installed.

As I understand there were fundamental changes to the underlying screen-controller communications I wanted to make sure that would not cause issues during a downgrade.

> The screen is not restricted for downgrading
@zauguin Interesting that the screen is unrestricted? I recall this is different than previously understood.

closed 2021-12-15 by brent113