Snapmaker/Snapmaker2-Controller #196: Firmware proposal

Matrixx200x, 2021-12-29:
https://forum.snapmaker.com/t/firmware-proposal/22771

1.) A screensaver for the control panel, as long as the prints run partially burns the standard image. A screensaver button or option would be advantageous.

2.) A function to change the tool, a button or function to move the axes and the tool head to the center (or freely selectable) of the Z and Y axis. To do this every time by hand is a bit cumbersome.

3.) An always visible light button for the housing light. If e.g. an error occurs e.g. filament empty or similar is no possibility to call the housing menu to turn on the light.
An always visible button that is always displayed above the screen would be great.

4.) Make the housing light intensity adjustable via a slider.

5.) Make the housing fan speed adjustable with a slider.

Greetings

Florian

From German Translated with DeepL

Cypress-L, 2022-01-06:
Hi, thanks for the suggestion, we will discuss it further internally
Here is a personal view on the five suggestions
1, screen protection function and adjust the fan speed function is currently in the design, to confirm a reasonable solution will be provided to you
2、The displacement function when changing the tool and the external cover light switch is a good suggestion, will be further discussed
3, about adjusting the intensity of the outer cover light, is the current light is too strong or too weak? Or in some scenes need to adjust the intensity of the light? Knowing these scenes helps us to bring better functional experience to everyone
Finally, thank you again for your suggestions, they are very practical, I will share your ideas to others and hope to implement it eventually

lionB777, 2023-07-28:
Hi,
Regarding the screen saver function, would you have an ETA ? I'm looking forward to have this since I received my printer after the KS campaign. 
Thanks.