Snapmaker/Snapmaker2-Controller #197: State messages via MQTT

spitzlbergerj, 2021-12-30:
It is quite difficult at the moment to query the current state of Snapmaker remotely. There are meanwhile various apps etc., but as far as I know they all use the Luban interface, with all the advantages and disadvantages of this API.
To be able to track the progress of 3D printing, CNC milling or laser cutting really independently, status messages via MQTT would be completely sufficient. A home automation system could use this data to display the status somewhere or also output status messages via Alexa and similar. MQTT would also allow processing via Node-Red etc. and would thereforebe very universally usable.
As status messages would offer itself:
at every change of line / at every 5th line output of the following information:
- current line
- total lines
- current time
- estimated time
- remaining time
- is filament out
- progress
- status

Sending MQTT messages should be configurable in the settings. For this purpose, the address of the MQTT broker, the port and, if necessary, the user and the password must be specified.

Eddie-Hartman, 2022-01-03:
I know this probably isn't exactly what you want, but just wanted to mention this: https://forum.snapmaker.com/t/snapmaker-status-app/20606/9

I made an app for monitoring the status of my machine for Android and posted it for free. Snapmaker Status if you want to look for it on the app store.

spitzlbergerj, 2022-01-06:
Hi @Eddie-Hartman,
I use your app extensively and with pleasure. Thanks for sharing.

However, I would like to output via Alexa for example that the 3d print is ready. I could also turn off the power to the printer once the printer is done e.g. sometime at night etc.

At least with your app I can check live if it's ready yet. That is very helpful. But it's not suitable for automating yet.

Unless you extend your app with the possibility to send MQTT telegrams :-) 

Regards
Josef

scyto, 2022-01-11:
MQTT would be great, in the mean time you could jury rig it using octoprint and home assistant (which has octoprint and can send messages to alexa).

[image]

yeah i know thats a lot of moving pieces and PITA if you dont already have ocotoprint and home assitant.

voidrin, 2022-01-15:
It will be more convenient if support MQTT, and I will add it into feature backlogs but the priority is not much high for now, we will improve remote connection performance first in touchscreen.

If any good ideas or recommended usage to use via MQTT please comments here.

And thanks @Eddie-Hartman for supporting us.

spitzlbergerj, 2022-01-15:
Thanks @voidrin for including this request in the backlog.

MQTT telegrams are especially helpful for state changes of the snapmaker. So if there are breaks of any kind in G-code, an MQTT telegram can trigger corresponding actions.

Examples for such state changes are:
- Error situations of any kind, where the SnapMaker is waiting for a user action.
- 3D printing: Filament run out
- 3D printing: pause to change filament
- 3D printing: pause to insert e.g. components into the 3-print like later inside ball bearings, screw sleeves, ...
- CNC: Pause to change the tool
- CNC: Pause to be able to suck off the object

A corresponding MQTT telegram can then e.g. trigger an alarm, so that such a condition can be quickly detected even if the SnapMaker is not in the same room.

Presumably, there are many more situations where the SnapMaker is waiting for an action from the user or where an action would be helpful. Then the MQTT Telegram can trigger appropriate actions or just inform the user that an action is necessary. 

The MQTT Telegram can then be used to ensure that a user triggers the actions that are best suited for him based on the installation situation that prevails at the user, such as that the printer is not in the same room as. 

Regards
Josef

spitzlbergerj, 2022-01-15:
> using octoprint and home assistant - thats a lot of moving pieces and PITA if you dont already have ocotoprint and home assitant.

I know octoprint, but I have not installed it because I am actually perfectly satisfied with the functionality that SnapMaker offers. I would really only need a few publiched states to work even more optimally. I have homematic as home automation and can also do a lot with it, if SnapMaker would publish a status on the net :-)

keesvanbemmel, 2022-01-15:
Just an "I'm experiencing this as well" post to show I would really like to see MQTT support 👯

scyto, 2022-01-19:
> I know octoprint, but I have not installed it because I am actually perfectly satisfied with the functionality that SnapMaker offers. I would really only need a few publiched states to work even [...]

i think you miss my point, asking for something here, it going inot backlog and then appearing is likely to take many months

i was just suggesting a workaround you can use in the interim - tbh if thats not interesting to do then i would suggest the feature isn't actually that important to you....