Snapmaker/Snapmaker2-Controller #199: Breaking Change For Monitoring Prints

Eddie-Hartman, 2022-01-03:
Hello.

I used to monitor prints that I'd start from the machine in Luban and a custom made app that would fetch the status from the machine. The latest firmware update seems to have a breaking change where once the machine is disconnected from Luban, it thinks it can no longer resume the current "engraving" (print). This behavior was observed with the 3d printing module.

Expected behavior: When connecting to a machine from Luban that has a print currently running, you can check status by sending a status request with a token api/v1/status?token={{token}} and get a response of the status. Once these requests are over being sent and the machine "disconnects" from Luban, the print should continue as if nothing ever happened.

Actual: The machine prompts the user to stop the current print because it disconnected and thinks it can not resume the print, even though it is printing from the file on the printer itself.
Photo: 
[image]

Side note but related: (I'm aware this isn't the Luban repo, but I figure the two teams must work closely)
In Luban itself, it is also demanding the machine go home before using the application. Again, you used to be able to connect without going home or stopping the current print. Clicking the OK button here just loops. It won't stop the current print. You can see the current correct status in the top right corner though, so it used to be fine for monitoring an in-progress print.
[image]

After clicking ok (the print is still running):
[image]

Eddie-Hartman, 2022-02-04:
@xhycode just wanted to bump this issue since it has been a month and I'm still running into this issue.
Any update?

I can track down the exact change that caused the issue tomorrow if that can help as a jumping off point then see if I can contribute to a change at all.

Cagiva-Raptor, 2022-02-04:
Indeed a long time :-(
I also have this issue when I want to connect a second laptop trough Luban when the print was started from Luban on the MacBook that sits next to the printer

Eddie-Hartman, 2022-02-05:
@parachvte @scotthsl @scyto
First of all, apologies for the series of @ s. I'm trying to get some visibility to this issue so it can get resolved quickly.

This is an issue that multiple people are running into and reporting it multiple times in the forum: https://forum.snapmaker.com/t/lost-connection-with-lubin/22984/2
Facebook: https://www.facebook.com/groups/snapmaker/posts/1395102530908056/
etc.

I'm here to try and help. I looked through the Snapmaker2-Controller repo and Luban repos and didn't find:

1. Any change that I could immediately address as the breaking change for this.
2. The message output on the screen text even searching multiple substrings.

So I'm guessing this change happened in some closed source repo that Snapmaker has, which in this case prevents me from helping to contribute to a fix. With that being said, if there is some movement behind the scenes to get this fixed, great, but can we have some communication from this end? If it hasn't been escalated to get fixed by Snapmaker, can we escalate it please?

If we can't connect to the machine and disconnect safely, then we can't connect to monitor status or have notifications hooked up without using some other controller like Octoprint. This is something many users want as reflected by these issues:
#215 #197 #181 #135 #38

I'm hoping for 2 possible things coming out of this:

1. The bug gets fixed.
2. Consider open-sourcing the code that this bug lived in. This would benefit us to resolve these kinds of issues faster, and can possibly add some visibility into the endpoints that Luban has available and the API requests that we can use to develop our own applications or notification systems. I'm obviously pushing this for personal reasons as I use this myself to monitor prints with my app, but I want it to benefit the Snapmaker Community. I really want to develop my own notification system for my phone, but don't want to risk putting a bunch of time and effort into a system that isn't open source and can have breaking changes like this happening with no response for over a month.

Eddie-Hartman, 2022-03-01:
I received an unfortunate update on this from a support ticket:

> Hi, Eddie,
> 
> Thanks for contacting us. We sincerely apologize for the late response.
> 
> This function works previously due to a bug.
Now, we have changed the interface logic. Once you discounted from Luban with a G-code ongoing, there will be a notification on the screen telling you that the machine is disconnected from Luban.
But if you do not press the OK button on the screen, the print will keep running.
If you press OK, the print will stop immediately.
You can leave the notification showing the screen when it is disconnected.
> 
> Have a nice day!
> Best regards,
> Zero

I disagree and responded as such in the support ticket. Posting here for visibility.

Martin-Visser, 2022-06-03:
I to have this problem, and it makes a filament change impossible after I connected to the wrong snapmaker. 

It is not good to have the interface you started the print with lock up after you connect with the software required to upload the files. 
I would even suggest that uploading the next file while still printing the current one is a highly wished for function.

Eddie-Hartman, 2022-07-09:
Any updates on this? It has been over 6 months.

loranger, 2022-09-01:
I'm interesting too having a readonly status monitoring api not requireing a (broken) touch validation

Eddie-Hartman, 2022-09-18:
I heard via Facebook from a snapmaker employee that this is being worked on and should get released in an update in November.

Cagiva-Raptor, 2022-09-18:
Awesome!Verstuurd vanaf mijn iPhone

Eddie-Hartman, 2022-12-18:
@xhycode or @sumibin any updates on this? November has come and gone and I've gotten no more updates via support tickets, the Facebook thread, or the forum. It's a pretty terrible user experience to have something get broken in an update for over a year with no real support responses.

Eddie-Hartman, 2023-02-05:
I feel like I'm shouting into a void, but any update here? Was told it was going to get fixed in November, then was told it would get fixed in the "next" firmware push, which came out recently, but no movement here and no mention of it in the release notes.

With the other issues that people are running into with the latest firmware like update loops and having to roll back, I'd rather not test it myself right now anyway. Also, the forum is currently in read-only mode.

FriedSemiconductors, 2023-02-05:
> I feel like I'm shouting into a void, but any update here? 

Yeah... kinda... I wrote two feature requests, that are not really critical, but come on... you won't get any answers at all.
They are both a year old, they assigned someone and then nothing.
No answer at all... no refusal ... no "yes" ....  no "maybe" ... not even if they are considering it.
It feels like it will stay in the status open indefinitely, like they parked it in a corner to never look at it again and many actual bug reports look like they are treated the same.

@Snapmaker
if you are announcing a new version of luban in your facebook groups, the comments below speak volumes ...
Your build quality feels mostly good, but many people don't trust the quality of your software.
You might want to adress that in the near future, since retaining customers is also important, not
just throwing new products on the market.

scyto, 2023-02-06:
@Eddie-Hartman  
I gave up, sold my Snapmaker A350 to someone who wanted to use it for laser and woodworking.

I never managed to get a good first layer over the whole device, it is fundamentally flawed IMO (speaking after 3 sets of rails, two 'frames' etc.

The flaw in the design requires incredible manufacturing precision, unfortunately the machine drifts to easily.  

tl;dr i am not going to be able to help you

bartipoo, 2023-08-25:
Has Snapmaker addressed this at all? Is there a way to see status of project running without being physically in front of the unit?
View from a PC on the same network? View from a mobile app? This would also let us get a notification if a print failed or a filament runout etc!

Eddie-Hartman, 2023-08-25:
@bartipoo no. They've given me the run-around so many times I got tired of asking. If you want to put another support ticket in you can try.

I literally wrote a free app to check statuses since they didn't make one and then they immediately broke it after. I could rewrite it to look a lot nicer and work on iOS, Android, Windows, etc, but there really isn't a point to doing that unless this bug gets fixed.

You still can technically check the status, but then if anything goes wrong you can no longer adjust anything on the machine without cancelling the print or messing with it over luban.

Eddie-Hartman, 2024-06-21:
Fixed with firmware update. Thank you.

closed 2024-06-21 by Eddie-Hartman

bartipoo, 2024-06-21:
Do you have the Release Notes info? I found this https://forum.snapmaker.com/t/snapmaker-2-0-firmware-updates-and-downloads/5443 v1.20.2 is he fix version correct?

Eddie-Hartman, 2024-06-21:
@bartipoo sorry I was just cleaning up old issues. I believe this was fixed in version V1.18.0.
> Refactor Touchscreen Print Job Management
In the previous version, when disconnected from Snapmaker Luban during print job, Touchscreen will request stop print job, and could not control the print job until reconnected from Luban.
In this version, when disconnected from Snapmaker Luban, Touchscreen will return to print. Print job control, power loss recovery and filament detection will recover on Touchscreen.

Here: https://forum.snapmaker.com/t/snapmaker-2-0-firmware-updates-and-downloads/5443/34