Snapmaker/Snapmaker2-Controller #201: Executing M502 on upgraded A350 sets wrong settings, risks damaging new v2 linear rails.

scyto, 2022-01-05:
thread for context https://forum.snapmaker.com/t/confused-on-settings-after-executing-m502/23293/3

summary:
I have A350, i upgraded the rails and head, firmwares auto updated, i noted that the numbers in M501 were different, assumed that was right because of upgraded rails and head.

I have been having odd printing issues (severe under extrusion on first layer) and i am struggling to know what changed because i have replace rails, head, hotend, filament batch, updated all my slicers, etc) so trying to go back a little.

I have put the old head back on, i reset from touchscreen and reset using M502. I am confused by output of M502 because its seems to have reverted to the old rails settings…

Before resetting with M502
```
Recv: echo:Steps per unit:
Recv: echo: M92 X160.00 Y160.00 Z400.00 B888.89 E228.18
Recv: echo:Maximum feedrates (units/s):
Recv: echo:  M203 X300.00 Y300.00 Z12.00 E120.00
Recv: echo:Maximum Acceleration (units/s2):
Recv: echo:  M201 X1000.00 Y1000.00 Z1000.00 E1000.00
```

after resetting with M502
```
Recv: echo:Steps per unit:
Recv: echo: M92 X400.00 Y400.00 Z400.00 B888.89 E212.21
Recv: echo:Maximum feedrates (units/s):
Recv: echo:  M203 X120.00 Y120.00 Z40.00 E25.00
Recv: echo:Maximum Acceleration (units/s2):
Recv: echo:  M201 X3000.00 Y3000.00 Z100.00 E10000.00
(i expected E212.21 as thats what a factory reset always did)
the other numbers are confusing me… my suspicion is M502 just did the wrong thing…
```

i am seriously not impressed with your culture of development and testing, you are flying fast and loose and continue to introduce issues that can potentially be catastrophic to the physical machine.  

You need to do better.

scyto, 2022-01-05:
reapplying firmware doesn't fix this

scyto, 2022-01-06:
please see this https://forum.snapmaker.com/t/confused-on-settings-after-executing-m502/23293/7?u=scyto  as such M502 should apply a reboot

xhycode, 2022-01-06:
Please help us get the Booting log for a further check. On the Console panel, input M1999 and press Enter. Then the machine will reboot and the Booting log will be displayed on the Console panel. Please copy the information and post it here. Thanks.

scyto, 2022-01-06:
is this everything?

would be good to know if everything looks good, having terrible time printing anything
not sure what it is that changed (new rails, new print head, filament batch, etc.

```
Send: M1999
Recv: will reboot machine
Recv: ok
[...]
Recv: start
Printer sent 'start' while already operational. External reset? Resetting line numbers to be on the safe side
Send: N0 M110 N0*125
Recv: echo:PowerUp
Recv: Marlin SM2-4.4.11
Recv:  echo:Compiled: Dec  6 2021
Recv: 
Recv: echo: Last Updated: 2021-12-06 | Author: Snapmaker Team
Recv: echo:Compiled: Dec  6 2021
Recv: echo: Free Memory: 24543  PlannerBufferBytes: 1792
Recv: set min_planner_speed:0.05
Recv: set min_planner_speed:0.05
Recv: echo:V73 stored settings retrieved (1077 bytes; crc 39401)
Recv: echo:  G21 ; (mm)
Recv: 
Recv: echo:Filament settings: Disabled
Recv: echo:  M200 D3.00
Recv: echo:  M200 D0
Recv: echo:Steps per unit:
Recv: echo: M92 X160.00 Y160.00 Z400.00 B888.89 E236.00
Recv: echo:Maximum feedrates (units/s):
Recv: echo:  M203 X120.00 Y120.00 Z40.00 E25.00
Recv: echo:Maximum Acceleration (units/s2):
Recv: echo:  M201 X3000.00 Y3000.00 Z100.00 E10000.00
Recv: echo:Acceleration (units/s2): P<print_accel> R<retract_accel> T<travel_accel>
Recv: echo:  M204 P1000.00 R1000.00 T1000.00
Recv: echo:Advanced: B<min_segment_time_us> S<min_feedrate> T<min_travel_feedrate> J<junc_dev>
Recv: echo:  M205 B20000.00 S0.00 T0.00 P0.05 L3.00 C0.05 J0.02
Recv: echo:Home offset:
Recv: echo:  M206 X-19.00 Y-10.00 Z0.00
Recv: echo:Auto Bed Leveling:
Recv: echo:  M420 S0 Z0.00
Recv: echo:  G29 W I0 J0 Z6.54375
Recv: echo:  G29 W I1 J0 Z6.49625
Recv: echo:  G29 W I2 J0 Z6.47125
Recv: echo:  G29 W I3 J0 Z6.47250
Recv: echo:  G29 W I4 J0 Z6.55125
Recv: echo:  G29 W I0 J1 Z6.65375
Recv: echo:  G29 W I1 J1 Z6.67250
Recv: echo:  G29 W I2 J1 Z6.63750
Recv: echo:  G29 W I3 J1 Z6.64375
Recv: echo:  G29 W I4 J1 Z6.67625
Recv: echo:  G29 W I0 J2 Z6.76875
Recv: echo:  G29 W I1 J2 Z6.92625
Recv: echo:  G29 W I2 J2 Z6.75250
Recv: echo:  G29 W I3 J2 Z6.79625
Recv: echo:  G29 W I4 J2 Z6.74000
Recv: echo:  G29 W I0 J3 Z6.80000
Recv: echo:  G29 W I1 J3 Z6.88000
Recv: echo:  G29 W I2 J3 Z6.87875
Recv: echo:  G29 W I3 J3 Z6.82250
Recv: echo:  G29 W I4 J3 Z6.79000
Recv: echo:  G29 W I0 J4 Z6.76750
Recv: echo:  G29 W I1 J4 Z6.77125
Recv: echo:  G29 W I2 J4 Z6.78250
Recv: echo:  G29 W I3 J4 Z6.74875
Recv: echo:  G29 W I4 J4 Z6.78125
Recv: echo:PID settings:
Recv: echo:  M301 P13.00 I0.10 D17.00
Recv: echo:Z-Probe Offset (mm):
Recv: echo:  M851 Z1.00
Recv: echo:Linear Advance:
Recv: echo:  M900 K0.04
Recv: Screen exists!
Recv: 
Recv: Message ID region:
Recv: emergent: 0 - 2
Recv: high    : 3 - 17
Recv: medium  : 18 - 53
Recv: low     : 54 - 127
Recv: 
Recv: Created marlin task!
Recv: Created HMI task!
Recv: Created heartbeat task!
Recv: Created can receiver task!
Recv: Created can event task!
Recv: Scanning modules ...
Recv: 
Recv: New Module: 0x20C7D9F8
Recv: Module 0x00C7D9F8: v1.11.4
Recv: 	Got axis X, endstop: 0
Recv: 	length: 356 mm, lead: 20 mm
Recv: 	Function [  0] <-> Message [  3]
Recv: 
Recv: New Module: 0x20C225C4
Recv: Module 0x00C225C4: v1.11.4
Recv: 	Got axis Z, endstop: 0
Recv: 	length: 356 mm, lead: 8 mm
Recv: 	Function [  0] <-> Message [  4]
Recv: 
Recv: New Module: 0x00507B00
Recv: Module 0x00507B00: v1.11.4
Recv: 	Got Enclosure!
Recv: 	Function [ 15] <-> Message [  5]
Recv: 	Function [ 18] <-> Message [ 18]
Recv: 	Function [ 19] <-> Message [ 19]
Recv: 
Recv: New Module: 0x2000BFE4
Recv: Module 0x0000BFE4: v1.11.4
Recv: 	Got toolhead 3DP!
Recv: 	Function [  8] <-> Message [ 20]
Recv: 	Function [  9] <-> Message [ 21]
Recv: 	Function [  6] <-> Message [ 22]
Recv: 	Function [  7] <-> Message [ 23]
Recv: 	Function [  1] <-> Message [  6]
Recv: 	Function [ 10] <-> Message [ 24]
Recv: 	Function [  2] <-> Message [  7]
Recv: 	Function [ 16] <-> Message [ 25]
Recv: 	probe: 0x1, filament: 0x1
Recv: set min_planner_speed:0.05
Recv: 
Recv: New Module: 0x20C22176
Recv: Module 0x00C22176: v1.11.4
Recv: 	Got axis Z, endstop: 0
Recv: 	length: 356 mm, lead: 8 mm
Recv: 	Function [  0] <-> Message [  8]
Recv: 
Recv: New Module: 0x20C7D930
Recv: Module 0x00C7D930: v1.11.4
Recv: 	Got axis Y, endstop: 0
Recv: 	length: 356 mm, lead: 20 mm
Recv: 	Function [  0] <-> Message [  9]
Recv: 
Recv: New Module: 0x20C7D9DE
Recv: Module 0x00C7D9DE: v1.11.4
Recv: 	Got axis Y, endstop: 0
Recv: 	length: 356 mm, lead: 20 mm
Recv: 	Function [  0] <-> Message [ 10]
Recv: axis index:0  pitch:160.00
Recv: axis index:1  pitch:160.00
Recv: axis index:2  pitch:400.00
Recv: Model: A350
Recv: grid manual
Recv: X:42 - 68
Recv: Y:28 - 76
Recv: disable door checking!
Recv: PL: first free block index: 14
Recv: PL: first non-free block index: 14
Recv: PL: no any data
Recv: PL: No data!
Recv: Finish init
Recv: 
Recv: SC req Exception
Recv: SC req enclosure sta
Recv: Eclosure: set LIGHT power 100
Recv: SC req emergency stop sta:1
Recv: SC req rotate sta:1
Recv: purifier info update timeout:0 - 2!
Recv: SC req purifier sta:1, err:0x0
Recv: purifier info update timeout:0 - 2!
Recv: SC req purifier sta:1, err:0x0
```

also typo, but no big deal, just FYI
```
Recv: Eclosure: set LIGHT power 100
```

xhycode, 2022-01-06:
No more problems from the boot message.
Is it possible that the machine works normally after rebooting?
M92 parameters default is X400 Y400 Z400, but your machine axis is X160 Y160 Z400
After restarting the machine, it will read the lead from the axis, so that it can move normally.
We did not notice this problem, sorry for your trouble!

scyto, 2022-01-06:
A reboot didn't seem to fix. Hard power reset was needed.

scyto, 2022-01-11:
@xhycode cool, does this PR fix this issue? https://github.com/Snapmaker/Snapmaker2-Controller/pull/202/files

xhycode, 2022-01-11:
Yes, this change will be included in upcoming releases

tHeSmUrF, 2022-04-30:
@xhycode has this been published yet? Like many people posting on the forum I've just had the same problem with the new modules grinding hard at the end of the rail.  I needed to do a reset on the settings to fix it. Not a good end user experience.

tHeSmUrF, 2022-05-02:
More to add to my last post. To fix the issue I manually reset the firmware settings as the machine wasn't detecting the new linear modules. This resolved my issue as it stopped grinding at the of the module.  As part of installing these I did ensure that I did a firmware check but I was already running the latest version and it reported no problems.  When installing the modules and testing them I did not have a tool head attached to the machine.  Later after attaching a brand new 3D Printer Head I got a message to say that the firmware on the tool head **AND** the linear modules was out of date and needed updating. 

I shouldn't have to plug in a tool head with an outdated firmware for the new linear modules to be detected.  If I had plugged in my old 3D Printer Head would the updated linear modules not been detected?