Snapmaker/Snapmaker2-Controller #208: When Z is at 324 X movement goes utterly screwy

scyto, 2022-01-10:
I have been using jog a lot.

I found:

-  if  i keep pressing up the max is  Z is 334.0
-  at Z = 334.00 pressing left right on X does very weird things deopending on speed chnages:

1. it might not move in either direction
2. it might move in one direction but will move in another direction
3. it might make really nasty sounds that scare me (like suddenly the stepping value being used went wild)

let me know if you need more info, for now i won't jog right to the top....

scyto, 2022-01-10:
oh its even worse than i thought - it breaks the internal coordinate system too
so whole coords on screen are right, issuing something like `G30 X200 Y200` actually moved the head to X = 320

scyto, 2022-01-10:
the jog issue doesn't exist if i:
```
G90
G1 Z 334
```
then X control is fine

scyto, 2022-01-10:
oh and this issue only occurs  intermittently - for example set to Z=334 with touchscreen
now home
then try moving X

scyto, 2022-01-10:
here is video to show what i mean, it will expire in two weeks
https://1drv.ms/v/s!AtFY20AHlltvjtVvcpRlwyzQvRB8uw?e=nNDuoq

xhycode, 2022-01-11:
I'm sorry for not reproducing the problem you described, I noticed that the SM2 hot bed is not used, will this problem still occur if the SM2 hot bed is used for leveling? Could you please provide the data of M402 V to help confirm if the problem is caused by leveling.

scyto, 2022-01-11:
I absolutely do have the SM2 hotbed installed. Not sure why you think that has anything to do with this?
This has nothing to do with leveling afaik. I had leveled with hot bed on. This test was after calibration.
Since then I have swapped to pei and re-leveled. Let me see if it still happens.

scyto, 2022-01-11:
Yes still have the issue.
Not sure you have understood - at Z = 334 the X left and right is not reliable. (334 is as high as my Z rail goes)
at z = 333 the X left and right is totally reliable

FWIW here is M420 V this time (this time is 1029 11x11 level, previous time was touchscreen heated 5x5 level)

```
Send: M420 V
Recv: Bilinear Leveling Grid:
Recv:       0      1      2      3      4      5      6      7      8      9     10
Recv:  0 -0.128 -0.177 -0.195 -0.218 -0.218 -0.214 -0.224 -0.224 -0.168 -0.102 -0.019
Recv:  1 +0.058 -0.031 -0.029 -0.115 -0.138 -0.114 -0.161 -0.134 -0.041 -0.048 +0.020
Recv:  2 -0.013 -0.016 -0.052 -0.090 -0.119 -0.120 -0.138 -0.130 -0.082 -0.010 +0.006
Recv:  3 -0.064 -0.056 -0.060 -0.122 -0.132 -0.135 -0.150 -0.170 -0.099 -0.056 -0.035
Recv:  4 -0.022 -0.009 +0.015 -0.059 -0.091 -0.055 -0.093 -0.109 -0.043 -0.034 -0.027
Recv:  5 +0.021 +0.039 +0.054 +0.044 -0.020 -0.045 -0.060 -0.025 -0.009 +0.000 +0.005
Recv:  6 +0.024 +0.056 +0.079 +0.040 +0.018 +0.016 -0.017 -0.026 +0.006 -0.011 -0.009
Recv:  7 +0.082 +0.119 +0.130 +0.071 +0.079 +0.091 +0.069 +0.021 +0.049 +0.051 +0.068
Recv:  8 +0.110 +0.157 +0.098 +0.079 +0.077 +0.082 +0.069 +0.056 +0.054 +0.081 +0.077
Recv:  9 +0.220 +0.153 +0.213 +0.151 +0.138 +0.194 +0.128 +0.138 +0.206 +0.143 +0.183
Recv: 10 +0.224 +0.198 +0.174 +0.160 +0.173 +0.166 +0.153 +0.164 +0.176 +0.195 +0.220
Recv: 
Recv: Subdivided with CATMULL ROM Leveling Grid:
Recv:         0        1        2        3        4        5        6        7        8        9       10       11       12       13       14       15       16       17       18       19       20       21       22       23       24       25       26       27       28       29       30
Recv:  0 -0.12750 -0.14537 -0.16324 -0.17750 -0.18555 -0.19000 -0.19500 -0.20296 -0.21148 -0.21750 -0.21931 -0.21861 -0.21750 -0.21602 -0.21412 -0.21375 -0.21644 -0.22065 -0.22375 -0.22657 -0.22829 -0.22375 -0.20949 -0.18898 -0.16750 -0.14718 -0.12588 -0.10250 -0.07597 -0.04736 -0.01875
Recv:  1 -0.05639 -0.08107 -0.10575 -0.12389 -0.12968 -0.12893 -0.13255 -0.14708 -0.16599 -0.18046 -0.18686 -0.18883 -0.18857 -0.18466 -0.17851 -0.17648 -0.18340 -0.19444 -0.20148 -0.20274 -0.20000 -0.19056 -0.16993 -0.14259 -0.11921 -0.10630 -0.09735 -0.08352 -0.06023 -0.03206 -0.00389
Recv:  2 +0.01472 -0.01677 -0.04825 -0.07028 -0.07380 -0.06786 -0.07009 -0.09121 -0.12049 -0.14343 -0.15442 -0.15906 -0.15963 -0.15330 -0.14290 -0.13921 -0.15036 -0.16822 -0.17921 -0.17890 -0.17172 -0.15736 -0.13037 -0.09620 -0.07093 -0.06543 -0.06882 -0.06454 -0.04449 -0.01676 +0.01097
Recv:  3 +0.05750 +0.02454 -0.00843 -0.03125 -0.03389 -0.02639 -0.02875 -0.05329 -0.08768 -0.11500 -0.12893 -0.13579 -0.13750 -0.13037 -0.11810 -0.11375 -0.12708 -0.14833 -0.16125 -0.16005 -0.15051 -0.13375 -0.10407 -0.06718 -0.04125 -0.03875 -0.04722 -0.04750 -0.03046 -0.00523 +0.02000
Recv:  4 +0.05236 +0.02792 +0.00349 -0.01449 -0.01922 -0.01749 -0.02319 -0.04525 -0.07474 -0.09880 -0.11297 -0.12172 -0.12551 -0.12066 -0.11085 -0.10764 -0.11952 -0.13799 -0.14912 -0.14789 -0.13932 -0.12449 -0.09915 -0.06755 -0.04352 -0.03558 -0.03520 -0.03060 -0.01644 +0.00194 +0.02032
Recv:  5 +0.01889 +0.00831 -0.00227 -0.01231 -0.02052 -0.02819 -0.03875 -0.05517 -0.07448 -0.09120 -0.10398 -0.11416 -0.12032 -0.11937 -0.11441 -0.11333 -0.12171 -0.13397 -0.14130 -0.14072 -0.13521 -0.12481 -0.10750 -0.08530 -0.06440 -0.04661 -0.03011 -0.01565 -0.00414 +0.00534 +0.01482
Recv:  6 -0.01250 -0.01255 -0.01259 -0.01625 -0.02588 -0.03912 -0.05250 -0.06523 -0.07810 -0.09000 -0.10125 -0.11153 -0.11875 -0.12060 -0.11940 -0.12000 -0.12556 -0.13292 -0.13750 -0.13833 -0.13639 -0.13000 -0.11806 -0.10167 -0.08250 -0.05810 -0.03093 -0.01000 -0.00042 +0.00292 +0.00625
Recv:  7 -0.03440 -0.03118 -0.02795 -0.02875 -0.03586 -0.04698 -0.05926 -0.07279 -0.08747 -0.10014 -0.10975 -0.11735 -0.12296 -0.12560 -0.12625 -0.12787 -0.13221 -0.13753 -0.14157 -0.14516 -0.14748 -0.14384 -0.13140 -0.11300 -0.09245 -0.06824 -0.04188 -0.02176 -0.01277 -0.01002 -0.00727
Recv:  8 -0.05421 -0.05105 -0.04789 -0.04736 -0.04990 -0.05508 -0.06421 -0.08048 -0.10070 -0.11667 -0.12452 -0.12811 -0.13079 -0.13311 -0.13452 -0.13671 -0.14052 -0.14510 -0.14968 -0.15678 -0.16387 -0.16255 -0.14696 -0.12296 -0.09963 -0.07884 -0.05873 -0.04282 -0.03369 -0.02877 -0.02384
Recv:  9 -0.06375 -0.06083 -0.05792 -0.05625 -0.05449 -0.05398 -0.06000 -0.07843 -0.10338 -0.12250 -0.13000 -0.13167 -0.13250 -0.13343 -0.13352 -0.13500 -0.13889 -0.14417 -0.15000 -0.15968 -0.16991 -0.17000 -0.15194 -0.12375 -0.09875 -0.08167 -0.06778 -0.05625 -0.04759 -0.04130 -0.03500
Recv: 10 -0.05694 -0.05349 -0.05003 -0.04690 -0.04145 -0.03632 -0.03977 -0.05947 -0.08775 -0.10981 -0.11960 -0.12317 -0.12393 -0.12085 -0.11495 -0.11278 -0.11772 -0.12638 -0.13509 -0.14623 -0.15742 -0.15792 -0.13899 -0.10937 -0.08454 -0.07077 -0.06179 -0.05426 -0.04755 -0.04229 -0.03704
Recv: 11 -0.03986 -0.03605 -0.03224 -0.02782 -0.01894 -0.00946 -0.01037 -0.03077 -0.06156 -0.08643 -0.09992 -0.10748 -0.10926 -0.10108 -0.08711 -0.08000 -0.08661 -0.10008 -0.11241 -0.12406 -0.13457 -0.13458 -0.11545 -0.08582 -0.06227 -0.05209 -0.04799 -0.04463 -0.04048 -0.03707 -0.03366
Recv: 12 -0.02250 -0.01829 -0.01407 -0.00875 +0.00204 +0.01394 +0.01500 -0.00389 -0.03361 -0.05875 -0.07518 -0.08704 -0.09125 -0.08153 -0.06417 -0.05500 -0.06282 -0.07884 -0.09250 -0.10255 -0.11023 -0.10875 -0.09065 -0.06338 -0.04250 -0.03523 -0.03435 -0.03375 -0.03148 -0.02949 -0.02750
[... cut ...]
Recv: 29 +0.22643 +0.20904 +0.19165 +0.18065 +0.18195 +0.18963 +0.19236 +0.18339 +0.16946 +0.15944 +0.15700 +0.15846 +0.16176 +0.16880 +0.17768 +0.18056 +0.17111 +0.15566 +0.14542 +0.14492 +0.14963 +0.15704 +0.16912 +0.18391 +0.19301 +0.18998 +0.18127 +0.17782 +0.18504 +0.19752 +0.21000
Recv: 30 +0.22375 +0.21491 +0.20607 +0.19750 +0.18903 +0.18083 +0.17375 +0.16745 +0.16227 +0.16000 +0.16292 +0.16875 +0.17250 +0.17208 +0.16958 +0.16625 +0.16130 +0.15551 +0.15250 +0.15435 +0.15898 +0.16375 +0.16759 +0.17157 +0.17625 +0.18181 +0.18806 +0.19500 +0.20287 +0.21144 +0.22000
Recv: 
Recv: leveling ON
Recv: echo:Bed Leveling ON
Recv: echo:Fade Height OFF
Recv: ok
```

scyto, 2022-01-11:
after doing that output above (which was done after inducing the fault) and then tapping home 3 times (it takes 3 taps to make it fully home after inducing the fault) the M420 V now looks like this

this was my bilinear measurement before inducing the fault - that's nasty this bug invalidate the fault!?

[image]

scyto, 2022-01-11:
doing hard power reset properly reset the M420 V - after the fault.
this is terrible, this bug totally invalidates co-oridinates!

```
Send: N1 M115*39
Recv: FIRMWARE_NAME:Marlin SM2-4.4.11 (Github) SOURCE_CODE_URL:https://github.com/whimsycwd/SnapmakerMarlin PROTOCOL_VERSION:1.0 MACHINE_TYPE:GD32F305VGT6 EXTRUDER_COUNT:1 UUID:cede2a2f-41a2-4748-9b12-c55c62f367ff
Recv: Cap:SERIAL_XON_XOFF:0
Recv: Cap:BINARY_FILE_TRANSFER:0
Recv: Cap:EEPROM:1
Recv: Cap:VOLUMETRIC:1
Recv: Cap:AUTOREPORT_TEMP:1
Recv: Cap:PROGRESS:0
Recv: Cap:PRINT_JOB:1
Recv: Cap:AUTOLEVEL:0
Recv: Cap:Z_PROBE:1
Recv: Cap:LEVELING_DATA:1
Recv: Cap:BUILD_PERCENT:0
Recv: Cap:SOFTWARE_POWER:0
Recv: Cap:TOGGLE_LIGHTS:0
Recv: Cap:CASE_LIGHT_BRIGHTNESS:0
Recv: Cap:EMERGENCY_PARSER:0
Recv: Cap:PROMPT_SUPPORT:0
Recv: Cap:AUTOREPORT_SD_STATUS:0
Recv: Cap:THERMAL_PROTECTION:1
Recv: Cap:MOTION_MODES:0
Recv: Cap:CHAMBER_TEMPERATURE:0
Recv: ok
Send: M21
Recv: echo:Unknown command: "M21"
Recv: ok
Send: M1010 S3 P100
Recv: Eclosure: set LIGHT power 100
Recv: ok
Send: G28
Recv: leveling OFF
Recv: leveling ON
Recv: Apply Z offset: 0.00
Recv: X:-19.00 Y:347.00 Z:327.50 E:0.00 Count X:0 Y:57120 Z:133600 B:0
Recv: ok
Send: M155 S2
Recv: ok
[...]
Recv: SC req Exception
Recv: SC req enclosure sta
Recv: Eclosure: set LIGHT power 100
Recv: SC req emergency stop sta:1
Recv: SC req rotate sta:1
Recv: purifier info update timeout:0 - 2!
Recv: SC req purifier sta:1, err:0x0
[...]
Recv: purifier info update timeout:0 - 2!
Recv: SC req purifier sta:1, err:0x0
[...]
Send: M420 V
Recv: Bilinear Leveling Grid:
Recv:       0      1      2      3      4      5      6      7      8      9     10
Recv:  0 +6.230 +6.180 +6.162 +6.140 +6.140 +6.144 +6.134 +6.134 +6.190 +6.255 +6.339
Recv:  1 +6.415 +6.326 +6.329 +6.242 +6.220 +6.244 +6.196 +6.224 +6.316 +6.310 +6.377
Recv:  2 +6.345 +6.341 +6.305 +6.267 +6.239 +6.237 +6.220 +6.227 +6.275 +6.347 +6.364
Recv:  3 +6.294 +6.301 +6.297 +6.235 +6.225 +6.222 +6.207 +6.187 +6.259 +6.301 +6.322
Recv:  4 +6.335 +6.349 +6.372 +6.299 +6.266 +6.302 +6.265 +6.249 +6.315 +6.324 +6.330
Recv:  5 +6.379 +6.396 +6.411 +6.401 +6.337 +6.312 +6.297 +6.332 +6.349 +6.357 +6.362
Recv:  6 +6.381 +6.414 +6.436 +6.397 +6.375 +6.374 +6.340 +6.331 +6.364 +6.346 +6.349
Recv:  7 +6.440 +6.476 +6.487 +6.429 +6.436 +6.449 +6.426 +6.379 +6.406 +6.409 +6.425
Recv:  8 +6.467 +6.515 +6.455 +6.436 +6.435 +6.440 +6.426 +6.414 +6.411 +6.439 +6.435
Recv:  9 +6.577 +6.510 +6.570 +6.509 +6.495 +6.551 +6.485 +6.495 +6.564 +6.500 +6.540
Recv: 10 +6.581 +6.555 +6.531 +6.517 +6.530 +6.524 +6.510 +6.521 +6.534 +6.552 +6.577
Recv: 
Recv: Subdivided with CATMULL ROM Leveling Grid:
Recv:         0        1        2        3        4        5        6        7        8        9       10       11       12       13       14       15       16       17       18       19       20       21       22       23       24       25       26       27       28       29       30
[... 12 lines cut ...]
Recv: 12 +6.33500 +6.33921 +6.34343 +6.34875 +6.35954 +6.37143 +6.37250 +6.35361 +6.32389 +6.29875 +6.28231 +6.27046 +6.26625 +6.27597 +6.29333 +6.30250 +6.29468 +6.27866 +6.26500 +6.25495 +6.24727 +6.24875 +6.26685 +6.29412 +6.31500 +6.32227 +6.32315 +6.32375 +6.32602 +6.32801 +6.33000
Recv: 13 +6.35093 +6.35555 +6.36017 +6.36569 +6.37560 +6.38641 +6.38861 +6.37579 +6.35435 +6.33398 +6.31587 +6.29884 +6.28903 +6.29283 +6.30386 +6.30912 +6.30115 +6.28742 +6.27731 +6.27334 +6.27298 +6.27815 +6.29289 +6.31315 +6.32861 +6.33431 +6.33520 +6.33583 +6.33763 +6.33916 +6.34069
Recv: 14 +6.36713 +6.37220 +6.37728 +6.38264 +6.39014 +6.39793 +6.40069 +6.39614 +6.38656 +6.37352 +6.35447 +6.33196 +6.31514 +6.30947 +6.30949 +6.30796 +6.30032 +6.29112 +6.28685 +6.29106 +6.30019 +6.31005 +6.32021 +6.33109 +6.33972 +6.34453 +6.34710 +6.34917 +6.35116 +6.35266 +6.35417
Recv: 15 +6.37875 +6.38468 +6.39060 +6.39625 +6.40236 +6.40819 +6.41125 +6.41176 +6.40949 +6.40125 +6.38255 +6.35787 +6.33750 +6.32593 +6.31866 +6.31250 +6.30491 +6.29842 +6.29750 +6.30616 +6.32037 +6.33250 +6.33958 +6.34458 +6.34875 +6.35236 +6.35514 +6.35750 +6.35945 +6.36097 +6.36250
Recv: 16 +6.38056 +6.38811 +6.39565 +6.40264 +6.40979 +6.41638 +6.41963 +6.41906 +6.41514 +6.40657 +6.39018 +6.36912 +6.35162 +6.34145 +6.33482 +6.32861 +6.32018 +6.31215 +6.30931 +6.31531 +6.32648 +6.33657 +6.34379 +6.34991 +6.35412 +6.35535 +6.35466 +6.35435 +6.35532 +6.35666 +6.35801
Recv: 17 +6.37778 +6.38735 +6.39692 +6.40569 +6.41490 +6.42331 +6.42648 +6.42161 +6.41150 +6.40009 +6.38728 +6.37316 +6.36199 +6.35678 +6.35452 +6.35042 +6.34087 +6.32948 +6.32222 +6.32182 +6.32556 +6.33120 +6.34005 +6.35080 +6.35741 +6.35595 +6.35033 +6.34620 +6.34594 +6.34716 +6.34838
Recv: 18 +6.38125 +6.39245 +6.40366 +6.41375 +6.42426 +6.43366 +6.43625 +6.42727 +6.41148 +6.39750 +6.38801 +6.38032 +6.37500 +6.37421 +6.37579 +6.37375 +6.36398 +6.35060 +6.34000 +6.33370 +6.33018 +6.33125 +6.34088 +6.35509 +6.36375 +6.36088 +6.35245 +6.34625 +6.34560 +6.34718 +6.34875
Recv: 19 +6.39782 +6.40970 +6.42158 +6.43213 +6.44308 +6.45271 +6.45449 +6.44222 +6.42210 +6.40620 +6.39942 +6.39686 +6.39597 +6.39784 +6.40139 +6.40083 +6.39259 +6.38024 +6.36870 +6.35775 +6.34761 +6.34393 +6.35251 +6.36754 +6.37727 +6.37519 +6.36781 +6.36282 +6.36360 +6.36678 +6.36995
Recv: 20 +6.42065 +6.43281 +6.44498 +6.45551 +6.46616 +6.47518 +6.47565 +6.46027 +6.43633 +6.41880 +6.41445 +6.41651 +6.41958 +6.42357 +6.42856 +6.42944 +6.42361 +6.41367 +6.40227 +6.38716 +6.37059 +6.36204 +6.36871 +6.38339 +6.39384  [... cut ...]
Recv: 29 +6.58393 +6.56654 +6.54915 +6.53815 +6.53945 +6.54713 +6.54986 +6.54089 +6.52696 +6.51694 +6.51450 +6.51596 +6.51926 +6.52630 +6.53517 +6.53806 +6.52861 +6.51316 +6.50292 +6.50243 +6.50713 +6.51454 +6.52662 +6.54140 +6.55051 +6.54748 +6.53877 +6.53532 +6.54254 +6.55502 +6.56750
Recv: 30 +6.58125 +6.57241 +6.56356 +6.55500 +6.54653 +6.53833 +6.53125 +6.52495 +6.51977 +6.51750 +6.52042 +6.52625 +6.53000 +6.52958 +6.52708 +6.52375 +6.51880 +6.51301 +6.51000 +6.51185 +6.51648 +6.52125 +6.52509 +6.52907 +6.53375 +6.53931 +6.54555 +6.55250 +6.56037 +6.56893 +6.57750
Recv: 
Recv: leveling ON
Recv: echo:Bed Leveling ON
Recv: echo:Fade Height OFF
Recv: ok
```

scyto, 2022-01-11:
i cant reproduce the mesh invalidation (the power cycle seems to have fixed that) i know this issued caused because i printed just fine before doing the Z=334 test

i can still create the failure at Z=334 where X left and right becomes totally wacky

scyto, 2022-01-11:
I am sorry, I don’t understand, what are you asking me to do?
Do you mean sensor when you say ‘actuating head’?

xhycode, 2022-01-11:
Please check if the 3D print head is installed correctly.

scyto, 2022-01-11:
3D print head **is and was** installed correctly (correctly aligned on the X axis mount).

Proximity sensor re-adjusted to be 1mm higher than nozzle tip using feeler gauge for accuracy to match default M851 (note a credit card in the us is 0.8mm thick, not sure if you should keep telling people to use credit cards).  It had changed relative to the hot end due to replacement of the hotend where the hotend height was slightly different.

I will do heated 5 x 5 auto calibration from touchscreen and see if I can reproduce.
I will do 1029  gcode 11x11 auto calibration and see if I can reproduce.

scyto, 2022-01-11:
Reproduced 100% using touchscreen.
M420 V as requested.
And because you are not believing me - here is a complete video (i saved you from having to watch most of the calibration part :-) )  https://youtu.be/xDbt3Rha_kU  - trust me when i say moving the X  when Z = 334 on this machine INVALIDATES MACHINE STATE  - and as i say this led to the M420 V mesh suddenly adjusting itself by -6.1mm for NO REASON i did nothing but what you see on the screen several times when the mesh invalidated itself - we have seen this mesh invalidation before - its IMO because the touchscreen doesn't issue gcode for control, it talks directly on the CAN bus.  

If I repeat the test using only gcode to get Z = 334 then the issue doesn't occur.  I suggest that all control from the touchscreen should be done via GCode - that way you would avoid this whole class of bugs.  
```
Send: M420 V
Recv: Bilinear Leveling Grid:
Recv:       0      1      2      3      4
Recv:  0 +6.009 +5.949 +5.941 +5.946 +6.065
Recv:  1 +6.175 +6.216 +6.204 +6.206 +6.214
Recv:  2 +6.289 +6.442 +6.327 +6.354 +6.259
Recv:  3 +6.339 +6.431 +6.464 +6.379 +6.306
Recv:  4 +6.306 +6.322 +6.370 +6.316 +6.321
Recv: 
Recv: Subdivided with CATMULL ROM Leveling Grid:
Recv:         0        1        2        3        4        5        6        7        8        9       10       11       12
Recv:  0 +6.00875 +5.98681 +5.96486 +5.94875 +5.94190 +5.94088 +5.94125 +5.93778 +5.93569 +5.94625 +5.97741 +6.02120 +6.06500
Recv:  1 +6.06611 +6.05644 +6.04677 +6.03944 +6.03571 +6.03431 +6.03389 +6.03141 +6.02991 +6.03708 +6.05841 +6.08842 +6.11843
Recv:  2 +6.12347 +6.12608 +6.12868 +6.13014 +6.12951 +6.12774 +6.12653 +6.12504 +6.12412 +6.12792 +6.13941 +6.15563 +6.17185
Recv:  3 +6.17500 +6.19074 +6.20648 +6.21625 +6.21551 +6.20880 +6.20375 +6.20329 +6.20449 +6.20625 +6.20838 +6.21106 +6.21375
Recv:  4 +6.21917 +6.25221 +6.28526 +6.30352 +6.29484 +6.27137 +6.25481 +6.25638 +6.26486 +6.26829 +6.26100 +6.24867 +6.23634
Recv:  5 +6.25750 +6.30871 +6.35993 +6.38620 +6.36636 +6.32157 +6.29046 +6.29409 +6.31139 +6.31782 +6.30159 +6.27450 +6.24741
Recv:  6 +6.28875 +6.34995 +6.41116 +6.44250 +6.41884 +6.36532 +6.32750 +6.33028 +6.34875 +6.35375 +6.33106 +6.29491 +6.25875
Recv:  7 +6.31319 +6.37049 +6.42779 +6.45995 +6.44751 +6.40994 +6.38051 +6.37647 +6.38058 +6.37440 +6.34833 +6.31197 +6.27560
Recv:  8 +6.33055 +6.37577 +6.42098 +6.45102 +6.45715 +6.44810 +6.43491 +6.42115 +6.40325 +6.38143 +6.35448 +6.32360 +6.29273
Recv:  9 +6.33875 +6.37181 +6.40486 +6.43125 +6.45088 +6.46384 +6.46375 +6.44366 +6.41051 +6.37875 +6.35366 +6.32995 +6.30625
Recv: 10 +6.33403 +6.35753 +6.38104 +6.40222 +6.42445 +6.44435 +6.44954 +6.42970 +6.39513 +6.36440 +6.34494 +6.32930 +6.31366
Recv: 11 +6.32014 +6.33402 +6.34790 +6.36236 +6.38211 +6.40243 +6.40977 +6.39355 +6.36435 +6.34032 +6.32925 +6.32335 +6.31745
Recv: 12 +6.30625 +6.31051 +6.31477 +6.32250 +6.33977 +6.36051 +6.37000 +6.35741 +6.33356 +6.31625 +6.31357 +6.31741 +6.32125
Recv: 
Recv: leveling ON
Recv: echo:Bed Leveling ON
Recv: echo:Fade Height OFF
Recv: ok
```

xhycode, 2022-01-11:
I'm sorry for the problem with the way I phrased it, no distrust of you, just trying to help you solve the problem
My guess is that what triggered this problem is that the leveling data collection is too small and has gone to negative values.
Due to the negative number the compensation stops the movement in the X-axis when it is out of range of the Z-axis .
Please use the M420 S0 to turn off the leveling and then move again. I guess it should be able to move normally, if so, it is caused by the leveling problem.
This problem is generally caused by a large change in the position of the proximity sensor of the hot bed or 3DP

scyto, 2022-01-11:
I think you miss my point.  

It doesn't matter what caused the machine to get like this, there should not be any scenario that causes this.  There should be no edge case where the machine state can be invalidated by jogging around.

I am telling you when it gets in this state not only does the moving the X not work (a minor inconvenience) this invalid state confused the machine to change, for no reason, a perfectly good mesh that if used would have physically managed the machine.

To be super super clear this is the sequence of events (because i am not sure you are getting the issue).

1. Calibrate machine - it produces a mesh with measurements in the 5.9mm to 6.4mm range
2.  print - all is great
3.  print a job with z =-0.1 - all is great
4. make issue happen (do the jogging you see), log it on github
5. go eat my dinner
6. snapmaker asks me to do M420 V
7. the mesh now reports as being in 0.02 to 0.28 range 
had i try to print with this and not have checked then the head would have smashed down into buld surface to get to -5mm lower than my build surface - do you understand what a catastrophe that wuld have been?

I changed nothing, the mesh decided to lower itself by over 5mm purely as a function of jogging around on the touchscreen

scyto, 2022-01-11:
aka at no time should jogging EVER allow me to cross the boundary into 'out of range for the Z axis' if 334 is not actually in range i should not be able to jog there....

xhycode, 2022-01-17:
I'm sorry for the late reply, I've found the problem, the exception was caused by the limit switch of Z-axis being triggered.
You are right, we should avoid this situation and make a good safety protection.
As the new module is under development, I can't fix this problem in time. Please avoid this situation in your use, sorry for the inconvenience.