Snapmaker/Snapmaker2-Controller #209: 3d printing Calibration grid isn't working

paauserrat, 2022-01-10:
It's easy to see that something is wrong. Printed this in PLA after doing a 25 point calibration. Some parts present nozzle to low, others nozzle to high, others slightly better and ok. Some filament isn't sticking while some is super sticky to bed.

[image]

[image]

[image]

[image]

[image]

[image]

paauserrat, 2022-01-12:
After doing a G-Code 11x11 calibration the results where much better.

`G28 ;Home
G1029 P11 ;Set auto-level mesh to 11x11
G1029 S ;Save auto-level mesh configuration
G1029 A ;Start auto-leveling`

Once the Snapmaker stopped taking measurements I used the Z up/down console in Luban because it allows movements in increments of 0.02mm

Then executed this second macro:

`Auto-Level Finish (set Z-offset first)
G1029 S ;Save settings
G1029 D0 ;Confirm auto-leveling is on.
M420 V ;Verify auto-level results
G28 ;Home`

And the result I got after running this and doing the same G-Code as before:
[image]

paauserrat, 2022-01-12:
So why don't you include bigger mesh calibration? We don't mind to have to wait 20 minutes for it to calibrate as long as results in overall better printing quality. I suggest adding 11 point calibration (and even more) from the display so we can do (1x1, 3x3, 5x5, 11x11).
It would be great if speed is incremented too so calibration is faster.

NomadicVoxel, 2022-07-05:
I'm surprised that your bed is in that bad condition, but I agree that having this change to the software would be quite useful.