Snapmaker/Snapmaker2-Controller #213: Probe placement causes leveling issues

LauranceP, 2022-01-13:
When doing an auto calibration the center of the sensor is placed on the very edge of the print bed. This reduces the sensitivity of the sensor, resulting in the printer thinking the bed is further away than it actually is (greater than 1.5mm on my A350).
This issue results in an incorrect subdivided leveling mesh, causes the print head hit the bed and in some cases pushes the print sheet off the bed when starting the print. 
Viewing the leveling grid shows the much lower front edge.
[image]
Probing the front edge (using this macro [ProbeMacro.txt](https://github.com/Snapmaker/Snapmaker2-Controller/files/7866624/ProbeMacro.txt)) you can see a significant increase in height with every millimeter increase in the Y coordinate until around 8mm from the original point. Results here [Probe results.txt](https://github.com/Snapmaker/Snapmaker2-Controller/files/7866631/Probe.results.txt).
[image]
A temporary fix it to shift the print sheet 10mm forward before calibrating the bed. This gives the sensor enough area to detect properly. 
[image]
I am not sure if the problem would be solved by simply moving the calibration mesh by 10mm or if this is a symptom of a bigger issue with the auto calibration.

scyto, 2022-01-19:
as someone with significant issues on calibration first layers, i have never seen this

1. are you sure your machine is built correctly?
2. are you sure its not a issue with the flexible build surface?
3. are you sure the sensor is the right distance above the nozzle height (0.88mm to 1mm)

for #2 try turning your build plate around, that will quickly eliminate that
for #1 we have seen some folks get things slightly out of place which moves the build surface backward, worth checking
for #3 follow the support article on adjusting the sensor.

also make sure you flexible build surface is correctly aligned it should be aligned with the front of the 3 bits that stick out of the heated plate and NOT pushed back

scyto, 2022-01-19:
> When doing an auto calibration the center of the sensor is placed on the very edge of the print bed.

tl;dr this is the bit that worries me, this is not the case on my machine.....

LauranceP, 2022-01-19:
Yep, removed and replaced the Y axis and we are good. Thank you Scyto!

closed 2022-01-19 by LauranceP