Snapmaker/Snapmaker2-Controller #218: Module cannot move after firmware upgrade v14.1 or v14.2

Vuego123, 2022-02-07:
Is anyone else having a problem with a moving the module after upgrading to v14.1 or v14.2 ?

After upgrading to v14.1 > system restarts, but none of the axis can move. Not via manually jogging and not via the homing function. I downgraded back to firmware v13.2 and everything works again.

A while later, I received the new 10W laser module and decided to upgrade to v14.2 > same problem occurs.

scotthsl, 2022-02-13:
Hi @Vuego123,
Thank you for your feedback, and I am sorry to reply you so late, I have given this question to our support team and try to make our test team to reproduce this issue.
Want to confirm if you still suffer this problem now?

Vuego123, 2022-02-13:
Yes, the problem is still occurring.

I have made 2 videos of my problem: 
- A video with firmware v14.2 > no movement at all
- A video with firmware v13.2 > all linear modules are working. I have disassembled the X axis and the platform so you can see that all rails are moving independently.

Links to video:
- https://user-images.githubusercontent.com/39829323/153753384-7a20a64b-b722-4dd9-a43b-c6d0dfe0522a.mp4
- https://user-images.githubusercontent.com/39829323/153753513-10b393da-56fd-4206-afbc-0610cbfeebda.mp4

All other settings, hardware setup and cabling... is the same in both videos. So only the firmware is different. 

My machine settings: 
- A350 with enclosure and emergency stop 
- No linear rails have been upgraded, so they all have the same pitch of 8mm.

I have been using my machine for 1 year without any problems.

I have done the following tests, without any luck:
- unplugged enclosure
- unplugged emergency button
- switched linear rails between X, Y and Z
- WIFI installation of v14.2

- USB installation of both v14.2 and v14.1
- switched between CNC, 3D print and laser modules
- When I downgrade back to v13.2 everything works, however, I can't use the 10W laser with v13.2.

Your support team can give me extra instructions to test things if that helps to find the root cause.
Please let me know if you need any logs from my machine.

I noticed other customers are complaining about the same problem (on facebook, github and snapmaker forum).

WennnnnT, 2022-02-14:
> 
Hello, may I ask whether there is a mixture of 2.5 generation and 2.0 generation modules in your machine? They are both 8 lead but not compatible. Unsupported on V1.14.2

Vuego123, 2022-02-14:
Where can I find the label information? 

I can't find a label that is physically present on the linear modules. Do I need to look in the logs or in a menu on the touch screen?

Vuego123, 2022-02-14:
Below you can find an extract of my bootlog. First the log states that it has found my X, Y and Z axis. Finally at the very bottom it says "Cannot detect any Linear Module!". That does not seem like it should be.

others < Scanning modules ...
others < New Module: 0x20395ABC
others < Module 0x00395ABC: v1.11.5
others <    Got axis Z, endstop: 0
others <    length: 356 mm, lead: 8 mm
others <    Function [  0] <-> Message [  3]
others < New Module: 0x2000E304
others < Module 0x0000E304: v1.11.5
others <    Got toolhead 3DP!
others <    Function [  8] <-> Message [ 18]
others <    Function [  9] <-> Message [ 19]
others <    Function [  6] <-> Message [ 20]
others <    Function [  7] <-> Message [ 21]
others <    Function [  1] <-> Message [  4]
others <    Function [ 10] <-> Message [ 22]
others <    Function [  2] <-> Message [  5]
others <    Function [ 16] <-> Message [ 23]
others <    probe: 0x1, filament: 0x1
others < set min_planner_speed:0.05
others < New Module: 0x00802FD2
others < Module 0x00802FD2: v1.11.5
others <    Got EmergencyStop Module!
others <    Function [ 20] <-> Message [  0]
others < New Module: 0x203854AA
others < Module 0x003854AA: v1.11.5
others <    Got axis X, endstop: 0
others <    length: 356 mm, lead: 8 mm
others <    Function [  0] <-> Message [  6]
others < New Module: 0x00507AA0
others < Module 0x00507AA0: v1.11.5
others <    Got Enclosure!
others <    Function [ 15] <-> Message [  7]
others <    Function [ 18] <-> Message [ 24]
others <    Function [ 19] <-> Message [ 25]
others < New Module: 0x20392112
others < Module 0x00392112: v1.11.5
others <    Got axis Y, endstop: 0
others <    length: 356 mm, lead: 8 mm
others <    Function [  0] <-> Message [  8]
others < New Module: 0x20395C48
others < Module 0x00395C48: v1.11.5
others <    Got axis Y, endstop: 0
others <    length: 356 mm, lead: 8 mm
others <    Function [  0] <-> Message [  9]
others < axis index:0  pitch:400.00
others < axis index:1  pitch:400.00
others < axis index:2  pitch:400.00
others < Model: unknow
others < grid manual
others < Cannot detect any Linear Module!

scotthsl, 2022-02-15:
Hi @Vuego123 ,
Thanks for your feedback, from your bootlog I found a missing z-axis, is the machine missing a z-axis every time when it boots under v1.14.2?
Also can you provide the bootlog for v1.13.2?

Thanks
Scott

Vuego123, 2022-02-15:
Bootlog for firmware v1.13.2 with exact same hardware setup:

M502
others < leveling OFF
echo:Hardcoded Default Settings Loaded
ok
M500
echo:Settings Stored (1064 bytes; crc 861)
ok
echo:Unknown command: "M1012"
M1999
others < will reboot machine
ok
echo:PowerUp
others <  echo:Compiled: Sep 13 2021
echo: Last Updated: 2021-09-13 | Author: Snapmaker Team
echo:Compiled: Sep 13 2021
echo: Free Memory: 24543  PlannerBufferBytes: 1792
echo:V71 stored settings retrieved (1064 bytes; crc 861)
echo:  G21 ; (mm)
echo:Filament settings: Disabled
echo:  M200 D3.00
echo:  M200 D0
echo:Steps per unit:
echo: M92 X400.00 Y400.00 Z400.00 E212.21
echo:Maximum feedrates (units/s):
echo:  M203 X120.00 Y120.00 Z40.00 E25.00
echo:Maximum Acceleration (units/s2):
echo:  M201 X3000.00 Y3000.00 Z100.00 E10000.00
echo:Acceleration (units/s2): P<print_accel> R<retract_accel> T<travel_accel>
echo:  M204 P1000.00 R1000.00 T1000.00
echo:Advanced: B<min_segment_time_us> S<min_feedrate> T<min_travel_feedrate> J<junc_dev>
echo:  M205 B20000.00 S0.00 T0.00 J0.02
echo:Home offset:
echo:  M206 X-19.00 Y-10.00 Z0.00
echo:Auto Bed Leveling:
echo:  M420 S0 Z0.00
echo:  G29 W I0 J0 Z9.00000
echo:  G29 W I1 J0 Z9.00000
echo:  G29 W I2 J0 Z9.00000
echo:  G29 W I0 J1 Z9.00000
echo:  G29 W I1 J1 Z9.00000
echo:  G29 W I2 J1 Z9.00000
echo:  G29 W I0 J2 Z9.00000
echo:  G29 W I1 J2 Z9.00000
echo:  G29 W I2 J2 Z9.00000
echo:PID settings:
echo:  M301 P13.00 I0.10 D17.00
echo:Z-Probe Offset (mm):
echo:  M851 Z1.00
echo:Linear Advance:
echo:  M900 K0.04
others < Screen exists!
others < Message ID region:
others < emergent: 0 - 2
others < high    : 3 - 17
others < medium  : 18 - 45
others < low     : 46 - 127
others < Created marlin task!
others < Created HMI task!
others < Created heartbeat task!
others < Created can receiver task!
others < Created can event task!
others < Scanning modules ...
others < New Module: 0x203854AA
others < Module 0x003854AA: v1.10.3
others <    Got axis X, endstop: 0
others <    length: 356 mm, lead: 8 mm
others <    Function [  0] <-> Message [  3]
others < New Module: 0x20395ABC
others < Module 0x00395ABC: v1.10.3
others <    Got axis Z, endstop: 0
others <    length: 356 mm, lead: 8 mm
others <    Function [  0] <-> Message [  4]
others < New Module: 0x00802FD2
others < Module 0x00802FD2: v1.11.5
others <    Got EmergencyStop Module!
others <    Function [ 20] <-> Message [  0]
others < New Module: 0x2000E304
others < Module 0x0000E304: v1.10.3
others <    Got toolhead 3DP!
others <    Function [  8] <-> Message [ 18]
others <    Function [  9] <-> Message [ 19]
others <    Function [  6] <-> Message [ 20]
others <    Function [  7] <-> Message [ 21]
others <    Function [  1] <-> Message [  5]
others <    Function [ 10] <-> Message [ 22]
others <    Function [  2] <-> Message [  6]
others <    Function [ 16] <-> Message [ 23]
others <    probe: 0x1, filament: 0x1
others < New Module: 0x00507AA0
others < Module 0x00507AA0: v1.10.3
others <    Got Enclosure!
others <    Function [ 15] <-> Message [  7]
others <    Function [ 18] <-> Message [ 24]
others <    Function [ 19] <-> Message [ 25]
others < New Module: 0x20392112
others < Module 0x00392112: v1.10.3
others <    Got axis Y, endstop: 0
others <    length: 356 mm, lead: 8 mm
others <    Function [  0] <-> Message [  8]
others < New Module: 0x20395C48
others < Module 0x00395C48: v1.10.3
others <    Got axis Y, endstop: 0
others <    length: 356 mm, lead: 8 mm
others <    Function [  0] <-> Message [  9]
others < axis index:0  pitch:400.00
others < axis index:1  pitch:400.00
others < axis index:2  pitch:400.00
others < Model: A350
others < grid manual
others < disable door checking!
others < PL: first free block index: 14
others < PL: first non-free block index: 14
others < PL: no any data
others < PL: No data!
others < Finish init
others < Detected error in sensor of Bed! temp: -15.00 / 0
others < new B target temp: 0
others < cannot stop in current status[1]

Vuego123, 2022-02-15:
1 linear rail module is always missing in the bootlog. 
However with firmware v1.13.2 that does not seem to be a problem.

scotthsl, 2022-02-16:
Hi @Vuego123 ,
Per your bootlogs from two version, I doubt that one of your Z-axes is not working properly, please contact our support team and they will assist you in troubleshooting and resolving the problem.

Thanks
Scott

Rolzad73, 2022-03-11:
I had re-setup my machine (A350 from the Kickstarter) about a month ago for 3D printing.  The first thing I did was upgrade the firmware to the latest (v1.14.2) since it had been so long since I had used the machine.

Printing has been going fine up until today, when the Z axis stopped moving up in the middle of the print, jamming the PLA into itself.
It then started moving up and down on Z axis about 5mm, in a bit of a fit.  I hit the power switch at that point (not E-stop).

After I restarted the machine, I could get no movement on the device, no homing, no jog. 
I then took apart the machine, leaving the rails installed, and plugged in, but X-axis separated to see if it was a hardware issue. No movement.
In my research I tried factory resetting and eventually came to this issue #218

I decided to downgrade the firmware to v1.13.2, and after rebooting the machine I could get homing and jogging again.
I am not going through the calibration, etc. I expect to be printing again in a few minutes.

I have no logs to compare to help with the issue, because collecting the logs was causing reboots on the device before I upgraded.

Ingo-on-Github, 2022-05-29:
Hi @Vuego123, could you resolve it? My 350 suddenly stopped working after I switched the 10W laser back to CNC. Had to downgrade from 14.2 (which shows the missing linear modules in the bootlog) to 13.2 and everything is fine again. Which somehow rules out a hardware issue. Going back to 14.1 and nothing moved again. Keen to hear how it played out for you?

karmavore, 2022-06-02:
I'm having this exact same problem with my kickstarter A250. No linear motors or extruder motor were working under 14.1 and 14.2.  I just reflashed 13.2 and its working fine again. This seems like a serious firmware issue.

Ingo-on-Github, 2022-06-03:
@karmavore I did received valuable guidance from the Snapmaker support team. The difference is that 14.2 will drop dead if one of the linear modules is failing. Whereas 13.2 will use all the linear modules still alive. In my case the cable of the right linear Z-module was lose (I guess I touched it as I vacuumed after a CNC job). The left linear module still worked and dragged the left module along. Once I was aware of this I could actually see the dragging somewhat stuttering movement. After I fixed the cable - fortunately both Z-linear modules were still working fine - 14.2 is happy again.

So I would argue, that 14.2 has a better protection of the linear modules than 13.2 as it prevents a single linear module working for both (which can't be good if happening for an extended time). It would be much better, if the firmware would call out the missing linear module to direct the user to the root cause. The support person will pass this suggestion on the the technical team and request an update for the FAQ. 

I hope this helps.

niksbha, 2022-06-10:
Hii having the same problem with a new machine. Model A250T. after startup I get a direct update for firmware 1.14.2. and then the thing happned.

Ingo-on-Github, 2022-06-10:
@niksbha: looks like one of your double-modules doesn't work. I hope checking the cables will help. Otherwise I would try to physically decouple the double modules (by removing the bed and the horizontal module and check who is moving under 13.2.

niksbha, 2022-06-10:
yes, it might be the problem...
currently, I have a downgrade firmware to 1.13.2 and start the machine. the auto home happens and the machine's z and x-axis move halfway then stop. also can't move any axis manually with the controller display toggle button.

suspect: Y-axis is not moved a once may be.

I don't know how to Bootlog to know if the axis is connected or not.

Ingo-on-Github, 2022-06-10:
@niksbha connect Luban via USB cable and fire a M1999 command in the console.

niksbha, 2022-06-10:
okay, thank you for your support.
The problem is almost solved by interchanging the dual cable splitter.
[image]

now all axis are moving by toggle, current firmware is 1.13.2

kubarbaz, 2022-08-09:
I'm having the very same problem. Do we still have this after 6 months?

karmavore, 2022-08-09:
Yes, I had my bed whenever both were plugged in, so for the y axis I had
one unplugged, so 14.2 would never run. After my x-axis started failing
last month, I just caved and bought the new upgraded linear actuator
bundle. They're so quiet it's ridiculous and very smooth movement.

mhd-321, 2022-10-25:
Any update on this issue other than downgrading? I'm facing the same problem...

karmavore, 2022-10-25:
I completely replaced the linear actuators with a fully new set from
Snapmaker (few hundred $$$) however I now have my complete original set
sitting unused. 3 or 4 of these should be completely fine if you want to
buy them from me. Venmo/offerup

Rolzad73, 2022-10-26:
> 

-reply to a reply-

I ended up doing the same thing. bought a new 2.0 set.  They are a bit different though that the Z rails (up/down) are different and need to be placed in their correct position or again nothing will work with no indication why.  It was my fault  though for thinking they were all the same.

In the end it works even though it cost $$. But I still do have my old 5 set (1 broken) that I THINK they could still be used for non-Z rails as replacements if anything goes wrong, but I haven't tested that theory.

In closing, it might be a good idea to have messaging or verbose logging that end-users can read easily to diagnose problems like this instead of just failing quietly as it does now.

Rolzad73, 2022-10-26:
> I completely replaced the linear actuators with a fully new set from Snapmaker (few hundred $$$) however I now have my complete original set sitting unused. 3 or 4 of these should be completely fine if you want to buy them from me. Venmo/offerup
> […](#)

-reply to a reply-

I ended up doing the same thing. bought a new 2.0 set. They are a bit different though that the Z rails (up/down) are different and need to be placed in their correct position or again nothing will work with no indication why. It was my fault though for thinking they were all the same.

In the end it works even though it cost $$. But I still do have my old 5 set (1 broken) that I THINK they could still be used for non-Z rails as replacements if anything goes wrong, but I haven't tested that theory.

In closing, it might be a good idea to have messaging or verbose logging that end-users can read easily to diagnose problems like this instead of just failing quietly as it does now.