Snapmaker/Snapmaker2-Controller #221: Laser staying on after job canceled

nivekmai, 2022-02-19:
Reproduced while running [my draft pull request](https://github.com/Snapmaker/Snapmaker2-Controller/pull/219) so this could be caused by my changes, but I suspect it's something else.

Reproduction steps:

1. Load [danger test.nc](https://drive.google.com/file/d/16l9NVLUWS89optZjKNvv-HqanWPlAvZX/view?usp=sharing) onto the machine
    - This is just some text and I post processed by swapping `M5` to `M5 I` and removing `M8`
    - [source lightburn file and unmodified gcode|attachment](https://drive.google.com/file/d/16s6-2MfL5PZGV7JGI-brDL-qe5IquYEp/view?usp=sharing)
2. Select "start" to start job setup (I didn't try going through files, just the big blue start on the main screen)
3. Use "auto level" (not sure this actually matters, I didn't try using manual level)
4. Start job
5. Immediately cancel job
    - I canceled the job before it got to the top of the first "T": 
[image]
    - I believe this means I canceled somewhere in the first moves of the first cut, likely line 17 or 18:
```
16  ; Layer Plywood cut
17  G1 Y5.05 F200 I S25.5
18  G1X-1.887
19  G1 Y0.676
20  G1X4.539
```

Result: Laser will stay on as the toolhead raises (until roughly 50mm above the work piece). When starting another job (using "start" on the main screen), the laser will turn on to the power setting it was at last (10%). The only way to turn it off is to go to "laser power" in "adjust settings" before starting the job. (Also, I saw some weird stuff where it was turning on during jog moves, but couldn't consistently reproduce that)

nivekmai, 2022-02-19:
Added some extra logging, it seems that we attempt to turn off the laser, but it's not actually turning off. There also appears to be some state that is lingering, since the laser comes back on at full power when starting the next job.

https://github.com/nivekmai/Snapmaker2-Controller/commit/ceea17bcbe0f248f9741b76d86bf247cceffd1f0

Video of me reproducing the bug with that extra logging: https://youtu.be/7TgBrdZWH8s

I think I'm not understanding how the laser is actually controlled. I expected that it was the setting of tim_pwm that should be enabling/disabling the laser, but there was clearly a command to set it to 0 and that didn't turn it off.

snapmaker-lixiaolong, 2022-02-28:
Hello, nivekmai，thank you for your feedback. We recommend that you use our official firmware release for comprarison testing.

nivekmai, 2022-03-17:
Hi @snapmaker-lixiaolong, as I stated, this is an issue on master, not on release. 

I've submitted a fix as well, I'm just wanting this to be merged into release before the next release build is made.