Snapmaker/Snapmaker2-Controller #224: 3D Printing calibration doesn't use the full bed

paauserrat, 2022-02-24:
I'm complaining about my Snapmaker 2.0 a250 not using the full bed when doing any type of calibration.

This applies to doing screen 3x3 or 7x7 but also to 11x11 custom mesh commands.

Here I upload a photo with the entire bed and a red square to where to calibration takes place. As can be seen, a lot of margins aren't used, so the full bed is not taken into account.

[image]

snapmaker-lixiaolong, 2022-02-28:
Hello paauserrat
Thank you for your feedback. 
Snapmaker 2.0 series using the Bilinear leveling, there is no need to probe the edges, the edges will be auto-completed by using extrapolation.
There is some deistance between the probe sensor and the hotend, the machine actually use probe sensor for detction, so you see the grid shifted a bit.

closed 2022-02-28 by snapmaker-lixiaolong

viper0078, 2022-03-05:
I have long felt that this has been a problem for me as well. There are areas on the right side and top of the bed where the level is not detected, so the rafts tend to peel off in those areas, causing warping of the printed material. I hope you will seriously consider the problems that users are having. I will describe the problem occurrence in a new thread.