Snapmaker/Snapmaker2-Controller #225: 3D Printing calibration doesn't use the full bed

viper0078, 2022-03-05:
Calibration of 3D printing is not performed on the edge of the bed.
This can result in very poor filament fusion at the edges of the bed.
The following pictures show where automatic leveling is performed for the four corners of the bed.

Left front
[image]

Right front
[image]

Back right
[image]

Back left
[image]

The points detected by the sensor in the above photo can be summarized in a single image as shown in the figure below.
[image]

The red line is the sensor’s detection line, which is a considerable distance from the edge of the bed.
The area 2-3 cm from the edge of the print bed has no print bed level detected.

Next, watch the following video.
https://www.youtube.com/watch?v=_0U7DKqpwYs

This is a video of interrupted printing of a raft. An 11x11 calibration is performed immediately before.
In the area at the back of the print bed, the filament moves just by touching it with a finger. The filament has not been fixed at all.
This is due to the fact that this is an area that is not subject to automatic level adjustment.

I use rafts for PLA and ABS printing, but the rafts fail to settle and cause warping in the printed material.
The smoothness of the print bed varies from machine to machine, so some may not experience this problem.

I think this is a very unfortunate state of affairs when the hardware is so good and has the wonderful feature of automatic level adjustment.

I hope this problem will be resolved soon.

pauserratgutierrez, 2022-03-05:
I have opened a similar request and SM developers closed it, but in my opinion, this (as well as ABL) must be revised and recoded because automatic bed leveling is not working.

#224

viper0078, 2022-03-06:
On the far side of the print bed, you can clearly see that the filament has not settled, as shown in the photo.
[image]

I am sure there are many other high priority projects for the development team. That said, I don't want this issue to be overlooked. I hope that the software will be improved to take advantage of the great hardware performance.

pauserratgutierrez, 2022-03-06:
In my opinion this should be a priority thing to look at. There are a lot of users that can't use more than few centimeters in the center of the bed. That's mainly because of the ABL is not working correctly. The auto bed leveling calibration procedure must be re-worked. It's unacceptable. If my bed isn't straight the ABL must correct it, and it isn't.

snapmaker-lixiaolong, 2022-03-07:
Hi, viper0078
Thank you for your feedback.
If you are using 2.0 bed, we strongly recommend that you set the tmperature of the bed to the temperature at which it will be printed before leveling. Because the 2.0 generation bed will have some deformation around it after heating, 2.5 generation bed is specially improved for this problem.

viper0078, 2022-03-07:
The photo above shows the printing bed left warm for about 30 minutes before leveling and printing. Warming up does not solve the problem. Obviously, the area about 3 cm at the back of the bed is having trouble with many failures.
The best thing to do is to have the firmware address this issue. I would think it would be easy to extend the area by simply changing the coordinates of the detection point.
Or, if there is a guarantee that the 2.5 generation beds have been improved and can solve the problem completely, please supply them. Not for free, of course, but please take into account that this is a response to a defect.

snapmaker-lixiaolong, 2022-03-07:
Hi，viper0078
We are very sorry for the problem you are experiencing. Maybe your machine does have a problem with the bed. For this problem, we have done internal tests on several A350 machines, and the results show that after heating and leveling, the edge of the hot bed is able to print well.
Here is a print of the flattened edge before heating
[image]
Here is a print of the flattened edge after heating
[image]
Due to the magnet at the edge, it will affect the detection sensor when detect edge.

viper0078, 2022-03-07:
I keep writing to you, but we haven't improved that way here.

I have confirmed that there is a magnet on the edge of the bed. We can detect the level between the magnets, right?
Can you consider implementing it separately like in 11x11 mode?
Like 11x11+Bed Edge Leveling.

mmack, 2023-03-11:
Same here on the Artisan!