Snapmaker/Snapmaker2-Controller #226: ABL Question ❓

pauserratgutierrez, 2022-03-06:
Hi, a lot of people are having issues with first layer adhesion on 3D Printing module with Snapmaker 2.0. While ABL should compensate for a warped and not straight bed, it isn't in some cases. My prints stick absolutely amazing in the center, I can use the machine to print small things and the quality is outstanding. But as I said, when a print uses more area than the center the nozzle is too high from the bed. So something eith ABL is not working. Maybe ABL is ok but the method used to compensate for Z heigh while printing is not well developed. I noted that while a print is ongoing the printer doesn't respond instantly to inputs from the screen. Example: if I adjust Z heigh or speed it doesn't take effect nor apply until some seconds have passed or the printer has ended a gcode line. So maybe live Z adjusting is not adjusting instantly as it should and doesn't compensate well.

I don't know what it is, but ABL and all first layer things should be adressed, because it is not normal.

snapmaker-lixiaolong, 2022-03-07:
Hi, pauserratgutierrez
Thank you for your feedback.
To the first issue, if you are using 2.0 bed, we strongly recommend that you set the tmperature of the bed to the temperature at which it will be printed before leveling. Because the 2.0 generation bed will have some deformation around it after heating, 2.5 generation bed is specially improved for this problem.
To the second issue, real-time print height adjustment is not yet available, and need to wait for the execution of the already planned motion before the print height is adjusted. But your suggestion is very good and we are considering to optimize this feature on the new firmware.

pauserratgutierrez, 2022-03-07:
> Hi, pauserratgutierrez
> Thank you for your feedback.
> To the first issue, if you are using 2.0 bed, we strongly recommend that you set the tmperature of the bed to the temperature at which it will be [...]

Hi, thanks for your fast reply.

Yes, I'm using 2.0 bed. I did some calibration tests with bed and nozzle at printing temperature. The problem is that it's not possible to do a 11x11 calibration from luban gcode with heated bed because it is automatically warmed by firmware. The screen allows to do bed leveling with heated bed.

How much time should I left the heated bed on before starting the print to make the bed go to its max deformation?

So if bed deformation increases in time till let's say 30 minutes (when deformation reaches the maximum), we should not print till bed has been heated for 30 minutes to achieve perfect conditions to 3d print, right?

In my opinion this can be fixed or at least mitigated a little using programation. Maybe the deformation is always the same depending on the time, so the firmware should calculate the amount of time passed and live ajust the distances automaticly depending to the deformation factor for time passed and print position.

If the 2.5 bed fixes the issue, you should allow us to at least have a discount because it is something we don't have blame of.

pauserratgutierrez, 2022-03-07:
> Hi, pauserratgutierrez
> Thank you for your feedback.
> To the first issue, if you are using 2.0 bed, we strongly recommend that you set the tmperature of the bed to the temperature at which it will be [...]

Oh, now I'm wondering... The 2.5 bed is the square one?  Or a newly developed one?