Snapmaker/Snapmaker2-Controller #233: G1029 A (Auto-leveling) Setting target temps  to 0

gerardov1, 2022-03-25:
I usually try to do my bed leveling with my bed at least at printing temp to get more accurate values as to consider expansion / contraction of the bed. I have been trying to set-up bed leveling from Octoprint (leveraging custom controls) and taking advantage in particular of the custom grid sizes provided by G-code G1029 P.

As I send the G1029 A command it seems the command also sets all target temps to 0...I consider this a bug as for leveling is important to have the ability to have the bed and may be even the extruder at a user defined temp.

Send: G1029 A
Recv: new E target temp: 0
Recv: new B target temp: 0
Recv: new E target temp: 0
Recv: leveling OFF

Can the behavior be changed to not automatically set temperatures to zero and respect the current temperatures?

Thanks!

snapmaker-lixiaolong, 2022-03-28:
Hi gerardov1
Thank you for your feedback. Previously, hotend heating and bed heating were turned off for safety reasons. We will submit your suggestion to our product manager, who will evaluate whether to remove this protection.

Matrixx200x, 2022-08-11:
Why has the problem not been fixed yet? The temperature is still set to 0.
It can't take that long to change the code.

Matrixx200x, 2023-02-03:
Push