Snapmaker/Snapmaker2-Controller #234: Please bring a FW that allows assembled the Controller on the other (left) Side

Voodoofly, 2022-04-04:
Hello Programmers,

mechanically all works to assembling the controller on the other (left) Side 
I flip the  X-Axis  too, you have to because the cable is too short

But now is the X-Axis reverse

Please make a option in the Firmware
I read in a forum you planed it for April 2021
Can you made it and make manny users happy please ? 

Controller standard, right Side 
Controller special feature , left Side

Many User wish this  : ) 

And the machine have more options for device installation

Anyone can loosen the touchpad's attachment themselves, thats not a problem

Thx
Tomislav

snapmaker-lixiaolong, 2022-04-11:
Hi, Tomislav
Thank you for your feedback. We have previously discussed your suggestion that such a change would require Luban and the screen to work with the change.

Zizzi34, 2024-03-21:
I follow. Need the same configuration firmware!