Snapmaker/Snapmaker2-Controller #236: Increase AutoLevel Mesh Size to 11

MikaAK, 2022-04-23:
A350 bed is to big for only a 5x5 leveling mesh, please increase to 7x7 at minimum or 11x11

snapmaker-lixiaolong, 2022-04-25:
Hi, MikaAK
Thankyou for your feedback. We should be adding leveling points in the near future.

MikaAK, 2022-04-25:
Amazing!! Thanks for the reply 😄

GregPeden, 2022-05-09:
Please also consider adding support for a sensor like BL Touch, as the magnetic technique I find to lack sufficient precision to be useful, additionally it only works with your OEM magnetic bed and nothing else.

MikaAK, 2022-05-16:
@GregPeden  create a new issue, adding onto them usually doesn't help

mmack, 2023-03-11:
I have a Artisan. Same Problem. The outer zones of the bed are not taken into leveling. Please allow that!