Snapmaker/Snapmaker2-Controller #237: Snapmaker 1.0, STOP/PAUSE function on touch panel does not work as can be reasonably expected.

TomaszSzt, 2022-05-04:
I am very surprised to see that STOP option at touch panel on the machine do await for current command to complete before stopping the machine. This is counter-intuitive, especially in CNC mode when user may expect stop to just stop the machine immediately.

The touch panel seems to behave exactly as a G-code "stop" command, but this is really not a right way to do it.

The fact, that "emergency stop" command do work immediately suggests that there should be no problem in implementing stop from touch panel to be a real immediate stop: stop all steppers, turn off laser and stop spindle, without however the "emergency" functionality.

A "Pause" function may be, in my opinion left as it is, it should however be some kind of information on a display like "Pause in progress...."

Request:

  Make STOP at touch panel to not wait for current command to complete.

TomaszSzt, 2022-05-04:
Hmm.... I suppose this is not the right place to report this problem, so please feel free to mark it as "not a valid request".

pauserratgutierrez, 2022-05-04:
I agree with your point. I think they have it that way so the emergency stop button addon makes sense...

snapmaker-lixiaolong, 2022-05-09:
Hi, TomaszSzt
Thankyou for feedback, we will submit your suggestion to our product manager.

TomaszSzt, 2022-05-12:
@[pauserratgutierrez](https://github.com/pauserratgutierrez)

The functionality of "pause" is very different from "emergency stop". After "pause" I can continue after removing the obstacle. After "emergency stop" I need to cycle power, re-connect to machine and can't continue. This is very different functionality.

Galindorf, 2022-07-03:
I have the enclosure with the A350, I also have this trouble with it taking much too long to pause or stop. If I want the machine to stop right away, I  pull open the door (I have door detection on with my enclosure so it automatically stops if I open the door)