Snapmaker/Snapmaker2-Controller #248: Snapmaker Losing calibration data on reboot.

pkdhital, 2022-07-07:
After recent firmware upgrade to Snapmaker2_V1.14.3_20220617, 3d printing calibration data are being lost on reboot. Here is console output when we calibrate machine, then enter M501, M503 and them M1999 to reboot machine. After that when we recheck calibration value, It's again at z9. Console output during this is attached.
[Snapmaker Dump.txt](https://github.com/Snapmaker/Snapmaker2-Controller/files/9060447/Snapmaker.Dump.txt)

snapmaker-lixiaolong, 2022-07-11:
Hi, it does appear from the logs that there is a problem as you say, but there is no further information in the logs to indicate why the problem is occurring. Our machine does not reproduce this problem, does this happen on your machine after each leveling is completed, if so please contact our after sales.

jdiggity81, 2022-09-04:
I have also experienced this.  So it is either that this is an issue with the controller as based on your advice for him to contact sales, or it is an issue with when updating the firmware when you m92 then m500 you will clear calibration data..

jdiggity81, 2022-09-04:
and mine isnt gone after reboot.  Calibration data is gone right after m500.

zhangquanyi, 2022-09-05:
> I have also experienced this. So it is either that this is an issue with the controller as based on your advice for him to contact sales, or it is an issue with when updating the firmware when you m92 then m500 you will clear calibration data..

I have checked the code of M92 and M500 and I don't find any problem for now. Can you provide the logs for the problem?

jdiggity81, 2022-09-05:
[image]

here is a pic.

zhangquanyi, 2022-09-05:
> [image]
> 
> here is a pic.

If the calibration data is detected to be reset, this pop-up window will appear on the screen, is this phenomenon after doing the operation of M92 and M500? If you can provide the logs at that time it would be appreciated.

jdiggity81, 2022-09-08:
So there are all the logs i could get off the machine.
It thinks i have a purifier which I don't. Says the laser is hooked up which it isn't. as why if the 3d printer module is on would it spit out anything about the laser level? I also don't have an enclosure.

1662390087303,2022.09.05 15:01:27.303,ERROR,SC-FW,purifier info update timeout:0 - 2!
1662390087304,2022.09.05 15:01:27.304,INFO,SC-FW,SC req purifier sta:1, err:0x0
37512,1970.01.01 00:00:37.512,INFO,SC-FW,SC req Exception
37577,1970.01.01 00:00:37.577,INFO,SC-FW,SC req enclosure sta
37581,1970.01.01 00:00:37.581,INFO,SC-FW,Eclosure: set LIGHT power 100
37587,1970.01.01 00:00:37.587,INFO,SC-FW,SC req emergency stop sta:1
37589,1970.01.01 00:00:37.589,INFO,SC-FW,SC req rotate sta:1
38094,1970.01.01 00:00:38.094,ERROR,SC-FW,purifier info update timeout:0 - 2!
38097,1970.01.01 00:00:38.097,INFO,SC-FW,SC req purifier sta:1, err:0x0
39079,1970.01.01 00:00:39.079,ERROR,SC-FW,purifier info update timeout:0 - 2!
39081,1970.01.01 00:00:39.081,INFO,SC-FW,SC req purifier sta:1, err:0x0
1662424187188,2022.09.06 00:29:47.188,INFO,SC-FW,SC req is leveled:1
1662424188019,2022.09.06 00:29:48.019,INFO,SC-FW,SC get env: 4
1662424188021,2022.09.06 00:29:48.021,INFO,SC-FW,live z offset: 0.050
1662424188062,2022.09.06 00:29:48.062,INFO,SC-FW,new laser power: -1.00
1662424188067,2022.09.06 00:29:48.067,INFO,SC-FW,SC req START work
1662424188082,2022.09.06 00:29:48.082,INFO,SC-FW,SC req -> Success

[SC_28.log](https://github.com/Snapmaker/Snapmaker2-Controller/files/9523367/SC_28.log)
[FW_0.log](https://github.com/Snapmaker/Snapmaker2-Controller/files/9523368/FW_0.log)
[SC_26.log](https://github.com/Snapmaker/Snapmaker2-Controller/files/9523369/SC_26.log)
[SC_27.log](https://github.com/Snapmaker/Snapmaker2-Controller/files/9523370/SC_27.log)

jdiggity81, 2022-09-09:
Now that i post the data we get no information.

zhangquanyi, 2022-09-10:
I'm very sorry for not replying to you in time.
thank you for the log file.
purifier is not connected, the screen tries to get the module status of the evolver and finally gets the timeout. This log is normal.
the settings of the enclosure and laser are the default operation of the screen. If the corresponding module is not connected, it will not have any effect.