Snapmaker/Snapmaker2-Controller #297: Dual Extruder: Z-Height Auto Calibration results in ~0.3mm too high Z-position

caerandir, 2023-05-23:
When I run Auto-Z-Calibration on the DX head (I *like* this feature!!!), since Firmware 1.15.19 the result is that the print starts about 0.3 mm too high, which I then need to compensate using the Live-Z-Offset. On the forum several other users report the same. I did a sensor calibration, but this did not help. It worked well before 1.15.19.

jenkins007, 2023-08-14:
I checked this also, and using firmware 1.16.7, mine was 0.3 mm too high. I couldn't use Luban as there appears to be no where to set a Z offset. I sent the file to the controller, adjusted the z offset printing locally by 0.3 mm and the print was fine. Hope this gets fixed soon! Seems like an easy fix...

jenkins007, 2023-08-19:
After reflection, I may have seen this in several previous firmware updates. For quite a few updates, the model would barely stick to the surface. Using the 0.3 mm offset down results in the model firmly sticking every time... So obviously something was amiss.

jenkins007, 2023-10-15:
I still have this issue using 1.17.17 firmware. 

If I use Luban or the controller, the first layer is “stringy” and has to be glued or removed. 
If I adjust settings on the controller and lower the Z offset -0.2 the print is excellent. This means that I no longer can use Luban to start the print jobs which was 90% of the time. Please Fix! 
Or have Luban to set a Z offset for all jobs.

yazz007, 2023-12-20:
I've encountered the same issue. I've had the Snapmaker2 for about a month now and I'm still getting familiar with it. However, I've consistently noticed problems with adhesion when printing using the right extruder. Initially, I suspected the Z offset might be the cause, but I wasn't entirely certain. I believed I wouldn't need to switch extruders if I intended to print using the filament loaded in the right one. In Luban, there's an option in the Workspace called 'Switch working nozzle.' I've tried using it, but regardless of the selection, it always prints using the left nozzle. I'm unclear about its intended function.
[image]

To get it to function properly, I edit the G-code file and specify printing on T1 instead of T0. However, the left extruder is at the same level as the right one, occasionally causing it to come into contact with the print and create a mess. It seems that T0 isn't lifting when it's not in use.

jenkins007, 2024-02-03:
Use the paint brush in Luban to "paint" your model with the right extruder selected. See if that helps and make is easier.
Looks like the offset has been fixed in the latest firmware updates. My model are not printing with very smooth base and somewhat difficult to remove, which is a very good thing... OK... Really not that difficult!

MIIB1, 2024-10-10:
Hello,
Same problem any solution?

Thank you