Snapmaker/Snapmaker2-Controller #299: G1029 A does not allow dual extruder?

silvanusbuyesti, 2023-06-07:
https://github.com/Snapmaker/Snapmaker2-Controller/blob/13e410d4bf97e15a855d378d2da6240db979e4a6/Marlin/src/gcode/bedlevel/abl/G1029.cpp#L114

The code is specifically disallowing this command from being executed against the dual extruder. Is there a plan to resolve this so we can do an 11x11 grid if we choose?

tylerhartwig, 2023-06-08:
Also running into this issue and looking for a solution.

caerandir, 2023-09-25:
Can confirm, and this is a problem. Have the lumpy kickstarter print bed and need 11x11 for large footprint prints.

dapendragon, 2024-11-08:
A fix for this would be greatly appreciated!

szimonek, 2024-12-21:
Still waiting (a year and a half) for this to be fixed.

nivekmai, 2025-01-17:
Ping @huang-w-y  @scotthsl @zauguin @xhycode is this ever going to get fixed?

zauguin, 2025-01-22:
@nivekmai Not sure what I have to do with this, but if you buy me a dual extruder I might consider contributing a fix.

chrisns, 2025-01-22:
looks like the code is here:
https://github.com/Snapmaker/Snapmaker2-Controller/blob/314c167b0a1ce331d867b10abd32e00908e4aa9c/snapmaker/src/service/bed_level.cpp#L692
that either needs to be referenced or called from the gcode or otherwise
Neither familiar enough with the codebase, know enough c, or brave enough to break my device to try faffing here 🤣

nivekmai, 2025-01-23:
> [@nivekmai](https://github.com/nivekmai) Not sure what I have to do with this, but if you buy me a dual extruder I might consider contributing a fix.

Sorry, I thought I saw you in the blame for dual extruder code, I guess I misread