Snapmaker/Snapmaker2-Controller #324: M600 Code has delay of some line

Verpohlt, 2024-04-08:
If I insert an M600 command into the G-code, it will only be executed by the printer 30-50 lines later.

MIIB1, 2024-10-10:
Hello,
Dud you solve the issue?

Thank you

Ray-Pfountz, 2024-12-26:
This worked for me. Use a text editor to edit the .gcode file. Go through and place M76 on it's own line at the lines of gcode where you need a pause. If you used PrusaSlicer and set a color change at the layers you needed, then it's simpler. Using the text editor, "Find" all of the M600 commands and place M76 on the next line, on it's own, after every M600 command or just replace the M600 with M76.  Either way, the print will pause and the toolhead will park. You must manually replace your filament then resume the print.
It was cool working the problem because it brought me back to my days of DOS 1.1 and programming in BASIC. (Beginners Allpurpose Symbolic Instruction Code) 1984 was a long time ago and George Orwell was running late.
I'm still trying to figure out how to use both of my extruders. PrusaSlicer is generating bad code when employing both extruders. It causes file parsing to fail. 
You will find this useful:
https://wiki.snapmaker.com/en/Snapmaker_Luban/manual/2_supported_gcode_references