Snapmaker/Snapmaker2-Controller #333: Host Action Commands

johnwc, 2024-12-02:
Does the latest firmware for A350 have HOST_ACTION_COMMANDS enabled/defined?

nivekmai, 2025-01-17:
No, it's commented out: 

https://github.com/Snapmaker/Snapmaker2-Controller/blob/314c167b0a1ce331d867b10abd32e00908e4aa9c/config/default/Configuration_adv.h#L2103

https://github.com/Snapmaker/Snapmaker2-Controller/blob/314c167b0a1ce331d867b10abd32e00908e4aa9c/Marlin/Configuration_adv.h#L2134

johnwc, 2025-01-19:
Can we get it enabled so third-party apps can receive information about prints, like when it was paused or stopped ,etc...

nivekmai, 2025-01-19:
I think the SM customizations are probably incompatible with it. They have their own system where they're keeping track of print status, and only issuing updates when the printer is in the proper state (and, as of now, that state is only updated when running prints from luban/the touch screen).

In #335 I made it so `M75` and `M77` update the print status in their custom tracking system, which then causes things like pause/stop to work. See https://forum.snapmaker.com/t/guide-octoprint-stock-filament-runout-detection/38751/4 where I detail how to set up Octoprint (and a custom firmware built with #335 ) to be able to handle filament runout.

johnwc, 2025-01-19:
Will this update to the firmware also cause the snapmaker UI show that it is printing when printing from octoprint?

nivekmai, 2025-01-20:
Oh, I haven't checked that, I actually have octoapp running on the touchscreen (https://forum.snapmaker.com/t/guide-how-to-install-apps-on-your-snapmaker-2-0-lcd-access-android-part-of-lcd/38614?u=nivekmai) so I've just been looking at octoprint and not using the snapmaker app anymore.