Snapmaker/Snapmaker2-Controller #338: GitHub sorely out of date

tgunr, 2025-04-03:
The last push to this repo was Dec 15, 2024???? 

Please update the repo so I can merge my changes.

ITmaze, 2025-04-03:
Unfortunately it appears that the Luban repository is in no better shape
and I'm beginning to wonder about compliance. Snapmaker doesn't appear to
even look at issues any longer, let alone respond.

I've had a ticket open since October last year because the Apple arm64
build is trying to execute x86_64, later found one for the same issue that
had been opened in February, 2024.

Neither issues appear to have had any response from Snapmaker, though I did
get a weird direct email (via GitHub, but not in the issue) from someone
claiming to be from Snapmaker that it was not reproducible.

In the 46 emails so far, Snapmaker Technical support says that the
developers cannot reproduce the issue, even with detailed instructions.

See:

   - https://github.com/Snapmaker/Luban/issues/2460
   - https://github.com/Snapmaker/Luban/issues/2513

I also can't seem to install it on Debian, with several missing
dependencies.

   - https://github.com/Snapmaker/Luban/issues/2575

I've not been able to build from source yet either.

tgunr, 2025-04-04:
I can build but only up to 4.5.34